abcminiuser/lufa #52: Issues with static libraries

Franky47, 2015-03-20:
Hello,

I'm trying to add LUFA to a firmware that is built using several modules, each built in a separate static library, and linked together to produce the final elf file, but I encounter a link error with `CALLBACK_USB_GetDescriptor`.
In order to keep it simple, I tried to replicate the project's structure using the ClassDriver MIDI Device demo in your repository, using my build system (based on CMake).

The whole project looks like this:
- LUFA, where only the sources from `LUFA/Drivers/USB` are compiled.
- lufa-demo-core, which contains the Descriptors and MIDI files from the demo, with the following modifications:
  - The `main` function has been removed from MIDI.c, and its contents have been spread out to two functions, `init` and `process` (one to be called before the loop, and the other one within). 
  - The appropriate declarations for `init` and `process` have been added in MIDI.h
- lufa-demo-firmware, which contains a single main.c file, implementing the `main` function calling `init` and `process` as described.

LUFA and lufa-demo-core each compile to liblufa.a and liblufa-demo-core.a, C static libraries.

lufa-demo-firmware links against these static libs to produce the final elf, but fails to find the `CALLBACK_USB_GetDescriptor` symbol, implemented in Descriptors.c, from its usage in `USB_Device_ProcessControlRequest` in the file DeviceStandardReq.c.

All LUFA components use the same LUFAConfig.h file (the one from the demo), and the signature for `CALLBACK_USB_GetDescriptor` matches in both the implementation and the call (so it's not a signature issue).

Please note that when building the demo files as-is in a single executable target, the link step succeeds.

I can't attach archives on GitHub, so here's a link to the modified demo files:
https://drive.google.com/file/d/0B7AbgN8zxRh0SmE0UDBoV0dWQTA/view?usp=sharing

More details if it helps:
- Building with avr-gcc 4.8.3 on Mac OS X 10.9.5, avr-libc version 1.8.1
- Building for ATmega32U4 (Arduino Leonardo), 16MHz for both F_CPU & F_USB.

Any insight would be most welcome !

Best regards,
Francois.

Franky47, 2015-03-20:
I just found that if I merge Descriptors and MIDI files together (in the MIDI file), it links alright. That's odd..

abcminiuser, 2015-11-08:
Oops. I'm guessing this is now solved (or you've long since given up and decided I was a fool) - please re-open if not and I'll help out ASAP.

closed 2015-11-08 by abcminiuser

Franky47, 2015-11-08:
Yep my "fix" in the second comment is fine enough for my use. Thanks for a very cool library !