abcminiuser/lufa #54: CDC Bootloader has no PROGMEM Device descriptor

NicoHood, 2015-07-22:
https://github.com/abcminiuser/lufa/blob/master/Projects/USBtoSerial/Descriptors.c#L46
vs
https://github.com/abcminiuser/lufa/blob/master/Bootloaders/CDC/Descriptors.c

Currently searching what has to be changed in order to fix this.

NicoHood, 2015-07-22:
What about something like this?
https://github.com/NicoHood/HoodLoader2/commit/6f1613aa8b8d7e5d42dbec7b5792e8636d20094b

NicoHood, 2015-07-22:
Also consider that <avr/io.h> contains the AVR Signatures
https://github.com/vancegroup-mirrors/avr-libc/blob/master/avr-libc/include/avr/iom32u4.h#L1497

https://github.com/NicoHood/HoodLoader2/commit/d88efba002c51a9aabf5a8db4fd5869165c701ca

abcminiuser, 2015-07-23:
You would need to edit the `Config/LUFAConfig.h` header of the bootloader, commenting out:
```
#define USE_RAM_DESCRIPTORS
```
And uncommenting:
```
#define USE_FLASH_DESCRIPTORS
```
To make it work. The rationale behind locating the descriptors in RAM was that the library uses the regular `pgm_read_byte()` function from avr-libc for the flash reads, which will fail on devices with >64KB of flash (AT90USB1287, AT90USB1286). To fix this I would have to add in a new configuration option to take 32-bit pointers for descriptors in flash, which seemed like too much complexity for a function only really needed in bootloaders where RAM was not scarce.

NicoHood, 2015-07-23:
I noticed that PROGMEM uses more flash (due to the read overhead, not the data itself) but less ram.
So for the 16u2 I use PROGMEM (because I need ram) and for any other MCU I use RAM. So in general I'd keep the setting like this, also for the reason you mentioned above.

If you uncomment USE_FLASH_DESCRIPTORS, do you still have to add the PROGMEM declarations?
You could at least add a note and some wrapper definitions like this:
https://github.com/NicoHood/HoodLoader2/blob/fast-usbserial/avr/bootloaders/HoodLoader2/Descriptors.c

NicoHood, 2015-07-23:
A quick question:
CDC_Device_LineEncodingChanged is called by
EVENT_USB_Device_ControlRequest is called by
USB_Device_ProcessControlRequest is called by
ISR(USB_COM_vect, ISR_BLOCK) or USB_USBTask()

Is this correct? The ISR is disabled by default. I am asking because I want to know if the LineEncoding will change randomly inside the code or only every loop. So is the USART setting save inside the loop or can it change?

NicoHood, 2015-07-25:
Is this really needed? In my modified bootloader version it works without this waiting. The Write function checks if the bank is not ready anyways and waits:
https://github.com/abcminiuser/lufa/blob/master/Bootloaders/CDC/BootloaderCDC.c#L662

NicoHood, 2015-08-30:
This could be improved this way that you dynamically use flash/ram for the descriptors. For you bootloader ram in general is better, because normally we have enough to use this and flash does not work on any mcu since it also adds more flash to the 4kb limit.

The point is that the information to find what I was looking for was somehow missing. It took some time to find out how to use flash descriptors.

closed 2015-08-30 by NicoHood