abcminiuser/lufa #55: Documentation Error: RAISE_EVENT macro does not exist

AndreasFMueller, 2015-07-24:
The description of the NO_LIMITED_CONTROLLER_CONNECT token states that on controllers with limited VBUS sensing capabilities, VBUS should be sensed by the application and the events raised by using the RAISE_EVENT macro. However, there is no RAISE_EVENT macro anywhere in the source code.

The documentation is also unclear about what should be done with the USB_DeviceState variable. Assuming I can figure out what state should be set when VBUS goes up (the documentation does not give a hint), it is still unclear how I get LUFA to do whatever is needed so that enumeration can start. Maybe this can work if the application calls USB_USBTask() regularly, but when INTERRUPT_CONTROL_ENDPOINT is set, then some interrupt must happen before any LUFA code can react to the change in USB_DeviceState. This looks like a documentation gap to me.

abcminiuser, 2015-07-25:
<what_year_is_it.gif>

Wowza, the `RAISE_EVENT` macro was from the first year or so of LUFA, back when it was a personal test project called "MyUSB". It's gone the way of the dodo, along with my silly pseudo-event abstraction macros in favor of regular functions for the events.

If I remember my implementation right, LUFA should cope just fine with `NO_LIMITED_CONTROLLER_CONNECT` defined on the U2 series AVR8 devices - the issue is that you would need to sense the VBUS connection and disconnection and fire `EVENT_USB_Device_Connect()` or `EVENT_USB_Device_Disconnect()` yourself to trigger any user code that relies on physical connection and disconnection. Internally, the stack will still figure out if it needs to re-enumerate or just wake up on an already enumerated host.

I'll update the documentation to clarify this.

AndreasFMueller, 2015-07-25:
I wonder how this could work, though. The documentation of the EVENT_USB_Device_Connect() function tells to do what you describe. It also says that USB_DeviceState should be changed manually. The default implementation of EVENT_USB_Device_Connect() ist empty (it's aliased to USB_Event_Stub), so it does nothing. And because no interrupt happens, no LUFA code can ever react to a manual state change in USB_DeviceState. Or should EVENT_USB_Device_Connect() have a different default implementation in this case?

abcminiuser, 2015-08-30:
Oops - didn't see the notification on GitHub. Sorry!

I had a look through my old code; you need to set `USB_DeviceState = DEVICE_STATE_Powered;` when your VBUS detection pin sees external VBUS applied, and `USB_DeviceState = DEVICE_STATE_Unattached;` when the VBUS is removed. You should also start/stop the internal PLL and raise the user events to ensure it works the same as the normal internal event:

**VBUS detected:**
```
			if (!(USB_Options & USB_OPT_MANUAL_PLL))
			{
				USB_PLL_On();
				while (!(USB_PLL_IsReady()));
			}

			USB_DeviceState = DEVICE_STATE_Powered;
			EVENT_USB_Device_Connect();
```

**VBUS removed:**
```
			if (!(USB_Options & USB_OPT_MANUAL_PLL))
			  USB_PLL_Off();

			USB_DeviceState = DEVICE_STATE_Unattached;
			EVENT_USB_Device_Disconnect();
```

closed 2015-11-01 by abcminiuser