abcminiuser/lufa #56: BootloaderDFU.c replies with wrong signature

Reuti, 2015-07-31:
Hi,

there are four bytes which should be output as signature:

30h Read Manufacturer Code
31h Read Family Code
60h Read Product Name
61h Read Product Revision

in BootloaderDFU.c I see (in todays Git-download):

<<<<
	const uint8_t SignatureInfo[4]  = {0x58, AVR_SIGNATURE_1, AVR_SIGNATURE_2, AVR_SIGNATURE_3};
	...
		  ResponseByte = SignatureInfo[DataIndexToRead - 0x60 + 3];
>>>>

Questions:

a) Isn't t the manufacture always 0x1E for Atmel, i.e. the AVR_SIGNATURE_1 value? What I see missing is the Product Revision where we might need an arbitrary value, but at the end of the list?

b) To cover access to the values which should be replied for the offset 0x60 and 0x61: maybe + 3 should read + 2 only. 0x61 would otherwise point behind the array when we add 3.

-- Reuti.

abcminiuser, 2015-11-08:
Hi Reuti, sorry I left this one dangling for so very long.

The manufacturer code isn't actually the 0x1E you expect; I can't remember where this is sourced from, but it's a differently allocated value to the regular Atmel manufacturer code that people know from the ISP signature (somewhat similar to the JTAG device manufacturer ID, which is different again). The 0x1E manufacturer is actually the first byte of the AVR's signature. This means that what the host sees for the AT90USB1287 is `0x58 0x1E 0x97 0x82`.

The access is indeed broken, and I very unwisely forgot the bounds checks. I've cleaned this up in https://github.com/abcminiuser/lufa/commit/6b06bc6237ccd63a268f82752d6b6c0265a77571 for the next release. Thanks!

closed 2015-11-08 by abcminiuser

Reuti, 2015-11-09:
Hi,

> Am 08.11.2015 um 04:51 schrieb Dean Camera <notifications@github.com>:
> 
> Hi Reuti, sorry I left this one dangling for so very long.
> 
> The manufacturer code isn't actually the 0x1E you expect; I can't remember where this is sourced from, but it's a differently allocated value to the regular Atmel manufacturer code that people know from the ISP signature (somewhat similar to the JTAG device manufacturer ID, which is different again). The 0x1E manufacturer is actually the first byte of the AVR's signature. This means that what the host sees for the AT90USB1287 is 0x58 0x1E 0x97 0x82.

Thx for the clarification - I wasn't aware of this.

> The access is indeed broken, and I very unwisely forgot the bounds checks. I've cleaned this up in 6b06bc6 for the next release. Thanks!

Perfectly solved.

Regards -- Reuti