abcminiuser/lufa #58: Wrong at90usb82 Signatures

NicoHood, 2015-08-08:
https://github.com/abcminiuser/lufa/blob/master/Bootloaders/DFU/Descriptors.h#L121
https://github.com/abcminiuser/lufa/blob/master/Bootloaders/CDC/Descriptors.h#L87

Might also be in different bootloaders.
Should be 0x93 instead of 0x94 (8kb flash, not 16).

Page 246 http://www.atmel.com/images/doc7707.pdf

Issue found here:
https://github.com/NicoHood/HoodLoader2/issues/6#issuecomment-128988794

```
avrdude.exe: Device signature = 0x1e9482
avrdude.exe: Expected signature for AT90USB82 is 1E 93 82
```

Also have a look at the issue filed for my HoodLoader2 (which I linked above):
CDC programming does not work for an 8u2 but with an at90usb82 (with the signature patched).
The 8u2 setting only works if you use avrdude with at90usb82 settings and the -F switch. So I think it is an avrdude issue, not lufa. Any ideas about that?

abcminiuser, 2015-08-30:
Looks like you are indeed very correct - I guess most people are using the ATMEGA8U2 rather than the older AT90USB82, so it hasn't come up until now. I'll patch it in the repo. Thanks!

abcminiuser, 2015-08-30:
Fixed in e10b5ffe94e1cea8aa9a97edf4a547fe00ae80ef.

closed 2015-08-30 by abcminiuser