abcminiuser/lufa #59: dfu-programmer eeprom command

untaugh, 2015-08-11:
The command for programming EEPROM memory with dfu-programmer has changed in newer versions. 

The one I have 0.6.2, uses 'flash-eeprom' instead of 'eeprom-flash'.

This still works in the latest version, but the recommended way appears to be 'flash --eeprom'

Also an erase command is needed before programming EEPROM (at least on atmega devices), which will leave the program memory blank.

abcminiuser, 2015-08-30:
Hrm; changing this would introduce an incompatibility with the older versions of dfu-programmer, although I suppose if 0.6.2 is old enough most people will be using it so a change won't impact many users.

Programming of EEPROM is deliberately done without an erase; only the written bytes will be affected leaving existing data intact. If the user wants to ensure the EEPROM is otherwise blank, the EEPROM HEX file should be 0xFF padded or the device fully erased before programming is attempted.

abcminiuser, 2015-11-08:
Fixed in https://github.com/abcminiuser/lufa/commit/4afebc8b70672e7a860606ba6e2496314265f1d9 - I had a look, and the command line arguments have been changed a long time ago, so it is unlikely people will not have migrated to a newer release.

closed 2015-11-08 by abcminiuser