abcminiuser/lufa #60: New release

NicoHood, 2015-09-02:
Its been a year since the last release, I suggest to create a new one.
There were a few bootloader fixes from my side which are very important. There are some opened issue left about that though. Not sure how long you want to wait for a new release, this should just be a reminder, that a year passed now since the last release.

abcminiuser, 2015-09-06:
It's a good point; with the last year or so I've had little time (and, let's face it, energy) to work on LUFA, as it's for the most part mature on the parts it was designed for (USB AVR8) and too ill suited for the more modern XMEGA for it to be worth extensive further development without redoing it from scratch with DMA/RAM mapped USB controllers in mind.

I've been waiting for a new "killer feature" before I packaged it up for a new release, but the number of changes over the last year has been small. I'm waiting for the final Atmel Studio 7 to drop before I recompile and package up the current LUFA master into AS7 and standalone form.

abcminiuser, 2015-09-27:
I'll be making a new release after with the new Atmel Studio 7 release, as it will require me to update and recompile the integration DLLs anyway -- so might as well use the opportunity to push out a new package to everyone. Shouldn't be long now, since they entered beta a few months ago.

NicoHood, 2015-09-28:
I'd like to update the cdc bootloader with a few tweaks and arduino compatibility. Are you willing to accept this? Then I could start working on that, but I dont know how long it will take and when I have time. Since the new release already added some CDC bootloader patches it would make sense to add this now.

Mostly it will add real Leonardo compatibility and the other features I added for HoodLoader2, except the serial bridge itself.

Also see my Uno board definition patch (PR).

Also if you still have a "project with lufa" page, could you please add HoodLoader2 to it?

abcminiuser, 2015-09-30:
Yes, I'll take a look over the weekend. There's no exact deadline for a new release, just when it's "done", when I have some free time, and when I've finished porting the integration binary to the new AS7 shell.

I'm doing the latter right now - there's some differences in the way packages are handled between the old and new Visual Studio Shell environments, and the differing resource DLLs will unfortunately require me to make separate AS6/AS7 versions of the extension.

The project page is generated from the LUFA documentation - I'll add a link for you so it's available along with the next release.

abcminiuser, 2015-11-01:
Integration work is completed (@xoriath managed to untangle the MSHelp mess for me) so I can now release whenever I want; I'll go through the issues first and see if there are any low hanging fruit first. Sigh - too much to do, waaaay too little time.

dhowland, 2015-11-12:
I'd love to see that AS7 release.

abcminiuser, 2015-11-13:
Would you mind installing this test release then please?

http://fourwalledcubicle.com/files/temp/LUFA-TESTING-15.11.12-20.57.21.vsix

In particular I want to check that the first run of AS7 after install (and only that first run) triggers the Getting Started page to show up automatically, and the integrated help installation prompt appears.

dhowland, 2015-11-13:
Installed.  The installation went smoothly, I've attached the log.

[VSIXInstaller_log_LUFA_testing.txt](https://github.com/abcminiuser/lufa/files/34058/VSIXInstaller_log_LUFA_testing.txt)

After the installer finished it didn't auto-launch AS7.  I launched it myself but unfortunately it did not auto-launch a getting started page or help installation.

I installed the help manually.  It worked, but it gave an unsigned warning:

[image]

Alright, started a sample project and started using the ASF wizard.  The extension shows up as version 00.00.00, so that may be a bug.  Next I had to agree to the license.  It successfully copied the files to my project.

So in conclusion it's usable but it seems like it isn't quite ready yet.

Thanks Dean, you do marvelous work.  I left a tip on 4WC

abcminiuser, 2015-11-14:
Thanks for testing.

> After the installer finished it didn't auto-launch AS7. I launched it myself but unfortunately it did not auto-launch a getting started page or help installation.

Aha! A bug! I've now patched that here:

https://github.com/abcminiuser/lufa-studio-integration/commit/b86d13df405b969cedb6b45782e188afc8b2a9fb

In previous versions of Atmel Studio (5.x and 6.x) the extension package class provided by Atmel Studio had a "first run" hook I used to use, so for that specific case I made the extension ignore the version information in the registry when determining if it is a new run after an updated installation. In AS7 they removed their overrides and switched to vanilla extension packages from the base Visual Studio SDK which omits this special hook, and I forgot to fix up the detection logic.

> The extension shows up as version 00.00.00, so that may be a bug.

That's OK, it's a "test release" which my scripts fill out with a dummy version number so I don't accidentally ship it to regular users. The real release will fix this up.

Now, I just need to fix up my PC so I can generate a final package for release to the Atmel Gallery; it seems the latest Win10 update breaks `sh.exe` which is crucial to all my packaging scripts.

abcminiuser, 2015-11-15:
Released as 151115 (see https://github.com/abcminiuser/lufa/releases/tag/LUFA-151115). Atmel Studio extension to follow shortly.

closed 2015-11-15 by abcminiuser