abcminiuser/lufa #63: AVRISP-MKII Clone non-function under Atmel Studio 7

abcminiuser, 2015-10-04:
There's been a payload minor revision change, but that alone doesn't fix the broken communication. I suspect there has been an extension to the protocol that I need to add to the existing clone firmware to make it compatible with AS7.

mattairtech, 2015-10-06:
It seems to work for me (AS7.0.582, Win7-64) with the modified version that I use for a programmer I sell. I was using the Jungo 11.1.0.0 (1/5/2013) driver. If this is not the latest driver, then the AS installation did not update it (and I don't see it in the device manager list of compatible drivers). I have to share timers in my implementation, so my timeouts are different. I also use a larger FIRMWARE_VERSION_MINOR (0x29).

420AmonRa, 2015-10-06:
OS Version: Microsoft Windows NT 6.1.7601 Service Pack 1 (win7-64)
Jungo 11.1.0.0 (05.01.2013).
FIRMWARE_VERSION_MINOR (0x29)
Atmel Studio 6 (Version: 6.2.1153) - all OK

Atmel Studio 7 (Version: 7.0.582) - don't work

Start AS 6 - now don't work too.

disconnect and connect AVRispmk2
Atmel Studio 6 (Version: 6.2.1153) - now OK

may be help to you.

mattairtech, 2015-10-08:
I tested the stock AVRISP MKII clone from the most recent Git and it too works on my system, so it doesn't *seem* to have anything to do with differences in timing with my fork. I tested this using an AT90USB646 at 16MHz with NO_VTARGET_DETECT defined, LIBUSB_DRIVER_COMPAT undefined, and FIRMWARE_VERSION_MINOR set to 0x29. I also set AUX_LINE_MASK to 0 (pin B0, also SS). The target board was an ATmega32U4.

abcminiuser, 2015-10-10:
Strange - definitely has communication issues on my old Win7 laptop with AS7 (worked with AS6.2) and my new Win10 PC with AS7 (haven't tried 6.2 on it yet). It might be interactions between the exact calls they make in the Jungo driver, and the way the Jungo driver talks to the USB backend in Windows. I've seen reports of failures under certain USB controllers, so who knows what's really going on.

avrguy, 2015-11-14:
Hi!
*First of all, sorry for my bad English.*
I have build an avrisp mkII clone, used your code with very minor modifications(just added some ctrl lines to select  ISP/PDI mode at hw level), and it's working flawlessly with Atmel Studio 6. However it doesn't works with Atmel Studio 6.2/7.0. Tested on Win8.1/Win10...

I found a patch on some russian site for Atmel Studio 6.2 that addresses this issue. I ported it to Atmel Studio 7.0 and guess what: it works on it too!
However there is an annoying bug(feature??) in AS 7.0: it resets the ISP clock to 125kHz every time it connects to the programmer; sends a CMD_SET_PARAMETER(PARAM_SCK_DURATION,6) command before CMD_GET_PARAMETER(PARAM_SCK_DURATION).

Here are the details(just patch a conditional jump):
**- for AS 6.2:** change the byte **0x74** to **0xEB** at file offset **0x35F11** in file \atbackend\codeCache\com_atmel_hil_usb.dll located in AS 6.2 install dir

**- for AS 7.0:**  change the byte **0x74** to **0xEB** at file offset **0x2F632** in file \atbackend\codeCache\com_atmel_hil_usb.dll located in AS 7.0 install dir

It is a temporary solution, but better than nothing.

DanKoloff, 2015-12-16:
I can confirm that the ISP clock gets reset to 125kHz.

I have a slightly different behavior, however, with Windows 10 and Atmel Studio 7.0.634:

While using "Device Programming" I click "Apply" to connect to the target and I get an error. Then if I click in the next second again "Apply" I would connect successfully (and I can read memories, program, check fuses etc). This happens repeatedly (when I disconnect using software means, for example, selecting "Simulator" and clicking "Apply"; then I attempt to select the programmer again I would again receive an error on the next "Apply" and if I click "Apply" in the next second or so I would get properly connected and so on).

Another strange thing is that if I use an USB hub (instead of connecting directly to the PC) I can never establish a connection.

I have commented out both  LIBUSB_DRIVER_COMPAT and RESET_TOGGLES_LIBUSB_COMPAT and I have bumped the firmware version. I use this also:

MCU          = at90usb162
ARCH         = AVR8
BOARD        = NONE
F_CPU        = 16000000

Just wanted to let you know for my experience. I believe it is related to some sort of USB timings or USB behavior that had been altered in Windows 10. I used Atmel Studio 7 fine under Windows 7.

abcminiuser, 2016-01-12:
Interesting, I've had other reports of success and failure. Both of my machines no longer work since I upgraded to Windows 10 and installed Atmel Studio 7, so I'd say it's down to a problematic new Jungo driver. I was told that a new version of Jungo shipped with AS7 to fix some installation issues, so perhaps some internal refactoring has caused the breakage.

ibelopuhov, 2016-01-22:
Hi everyone! I have devboard with at90usb162, and yesterday I faced with the problem. Win10 + AS7.0 does not work together.
The problem with the Jungo driver which has to be replaced with lib-usb32.
If someone interesting in resolving issue I can post detailed instruction how to compile mkII firmware and replace Jungo driver. The original article is here http://habrahabr.ru/post/273881

abcminiuser, 2016-01-27:
Hi everyone,

So, it turns out this is very likely a Jungo change that has broken compatibility with the clone. Atmel Studio 7 however ships with a new alternative backend for the AVRISP-MKII clone, using libUSB (as @ibelopuhov mentions above). However, Atmel Studio doesn't currently ship any signed libUSB drivers for this device, so while the capability is there users with versions of Windows that enforce driver signing won't be able to actually use it.

I have now tested a signed copy of the libUSB driver, replacing the Jungo driver, which appears to work fine on my test Win10 system. However, I can't release said driver as it is signed with a third-party certificate that cannot be used in an unsanctioned software release.

Therefore I'm crowdsourcing this: is there someone with a valid driver signing certificate who would be able to signed a generated libUSB driver for general distribution?

- Dean

mattairtech, 2016-01-27:
I have a code signing certificate that expires on Thursday, January 28. Send me the .inf and .cat (I can generate the .cat) and sign. Bear in mind that my company name, MattairTech LLC will be displayed when the user installs the driver.

abcminiuser, 2016-01-28:
The tested driver (actually it's just a stock libusb-Win32 driver with the right VID/PID set) is here:

http://fourwalledcubicle.com/files/temp/AVRISP_mkII-libusb0.zip

I've stripped out the existing signature CAT file, since it's not authorized for distribution. If you are able to signed it (again) for general distribution that would be *very* much appreciated. Does the cert expiring invalidate the signature, or just prevent you from re-signing it later on?

mattairtech, 2016-01-28:
I'm really sorry Dean, but my certificate expired on the 26th, not the 28th. I had it stuck in my head that it was the 28th. I might need to get another certificate. If I do, I will let you know. To answer your question, the cert expiring just prevents signing after the expiration date (the driver will still load).

mitchjs, 2016-01-28:
ok, I looked at the .sys file the actual driver, that is signed... assuming win10 still accepting a SHA1... which I think it is since it was signed in 2012
so its just Cat file... I just renewed my CERT... I always hate doing that!

I signed it... dean can I get It to you for a test?

ehajo, 2016-01-29:
I just signed the driver. Have a look here:
http://www.ehajo.de/AVRISP_mkII-libusb0.zip

I justed tried it on a fresh win10 installation and it works. For everyone: Feel free to use it!

abcminiuser, 2016-01-31:
Wonderful! @ehajo, would you mind if I included your signed versions of the USB2Serial, AVRISP and (possibly) RNDIS in the next release? If so, what notices or messages would you like me to include in the tree/manual?

@mitchjs and @mattairtech : thanks for the offers also - seems a lot of people are really willing to pitch in, which is really awesome. I actually now have an Atmel signed one too which may be releasable; I'm waiting on confirmation that it can be distributed publicly before I commit it. I think I should probably use the Atmel signed version if possible, but I'll still need signed serial and RNDIS INF files from the community.

Again, giant honking great big thanks to everyone for your continued interest in LUFA, your investigations and your offers to help.

abcminiuser, 2016-02-07:
Ok, getting back to this (no time, getting married in a few weeks!) I've done a quick run-through and it seems that the new signed Windows driver works with both Atmel Studio 7, and AVRDUDE 6.2. It also appears to work with AVRDUDE 6.1 on Linux, without the compatibility mode.

Thus: as much as I like backwards compatibility, it's now a burden I want to get rid of. As a result, I'm planning to remove the `LIBUSB_COMPAT_MODE` and `RESET_TOGGLES_LIBUSB_COMPAT` compile options, and scrub it from the docs. The firmware will only have the one endpoint layout, which now works with the new driver on Windows (Atmel Studio 7, AVRDUDE) and Linux (AVRDUDE). Other combinations won't be tested, but may work.

Anyone have any severe objections to this? Those really, really wanting the compat modes can still checkout an older tag and rebuild.

abcminiuser, 2016-02-07:
Alrighty, effectively fixed in:
6b0c46af6e4e67285fada7a3212197e92a33eb26 - Added signed driver
1d2369e9be57ceee8ba415655ce5f13d9ae8da84 - Removed legacy compat options

@ehajo for the programmer, I've gone with a Atmel signed version, just in case people don't trust an unknown (but respected!) small company. I'll still need your/someone's help with the other drivers, I'll make a new issue to track that.

Thanks everyone!

closed 2016-02-07 by abcminiuser

ehajo, 2016-02-07:
Full ACK with you. Remove the old stuff is a good idea-------- Originalnachricht --------Betreff: Re: [lufa] AVRISP-MKII Clone non-functional under Atmel Studio 7 (#63)Von: Dean Camera An: abcminiuser/lufa Cc: Hannes Jochriem Ok, getting back to this (no time, getting married in a few weeks!) I've done a quick run-through and it seems that the new signed Windows driver works with both Atmel Studio 7, and AVRDUDE 6.2. It also appears to work with AVRDUDE 6.1 on Linux, without the compatibility mode.

Thus: as much as I like backwards compatibility, it's now a burden I want to get rid of. As a result, I'm planning to remove the LIBUSB_COMPAT_MODE and RESET_TOGGLES_LIBUSB_COMPAT compile options, and scrub it from the docs. The firmware will only have the one endpoint layout, which now works with the new driver on Windows (Atmel Studio 7, AVRDUDE) and Linux (AVRDUDE). Other combinations won't be tested, but may work.

Anyone have any severe objections to this? Those really, really wanting the compat modes can still checkout an older tag and rebuild.

MNdiaz, 2016-06-16:
tengo un problema al cargar la LUFA PATH= 
this is the errror:
 Severity	Code	Description	Project	File	Line
Error		../../../lufa/makefile/LUFA/: No such file or directory	VirtualSerial	C:\Users\VAIO\Documents\Atmel Studio\7.0\USB_CDC\VirtualSerial\makefile	130
Could anyone help me please!!! thanks!

MNdiaz, 2016-06-16:
Hi I'm working with atmel studio 7 with the USB communication but I have an error when compiling the project, this is the error:

  Severity Code Description Project File Line
../../../lufa/makefile/LUFA/ Error: No such file or directory VirtualSerial C: \ Users \ VAIO \ Documents \ Atmel Studio \ 7.0 \ USB_CDC \ VirtualSerial \ makefile 130

anyone knows how to eliminate the error!

abcminiuser, 2016-06-30:
(Hurriedly typed on my phone - currently away on my European honeymoon)

That looks like a general issue with compiling under ATMEL studio. Are you
using the ASF extension of LUFA? If so it should work out of the box unless
there's a misconfiguration in AS7.

Do any other projects cause similar issues when compiled, either a new bare
C project or another ASF example?

- Dean

MNdiaz, 2016-07-06:
Hi Dean, thanks for the answers, I am very grateful to you for helping me, have made progress in the LUFA configuration, thanks to your help, but I have a new problem, I explain:
I worked with Atmel Studio 7, download the LUFA libraries that made me missing:

[cid:8dfb8c4a-83ae-48ea-b292-0dd59bac75c1]

then opened the next project for USB communication:

[cid:63aa9bd9-3967-4013-b3ee-9e52d32aa39d]

[cid:c3fa6a1a-d5ad-488b-8fbb-744c073795c6]

The problem is that when you compile the project I get the following error:
[cid:322bc003-441c-4c5d-a80f-e21eeae11982]

It is my mistake, when I connect the Arduino Leonardo in the device manager does not mark error, but after 5 seconds to load the bootloader ignores the device as a host, you think is the problem of Arduino Leonardo, or I need to download libraries USB, or I have to set the host on the program.

thanks for your help, good day.

abcminiuser, 2016-07-11:
Looks like your images were removed; can you host them elsewhere such as imgur?

Your previous message mentioned a makefile - that shouldn't be present in Atmel Studio when using the LUFA ASF extension, as it uses the native Atmel Studio build system instead. Did you manually create an AS project from the standalone LUFA release demos?

MNdiaz, 2016-07-20:
Hi, thanks for reply to the message,
If, when I started with communication via USB, initially try creating the makefile, but I think it is better through demonstrations LUFA, I hope to be right, a question, you must download libraries to recognize ArduinoLeonardo as host ?,
  or the computer must be the host and ArdinoLeonardo as device?
Thanks for everything, Good Day!

abcminiuser, 2016-07-31:
You will need to install Atmel Studio 7, along with the Atmel Software Framework extension (this should be an option in the installer) and the LUFA Atmel Studio extension (use the Extension Manager from inside Atmel Studio 7).

With both extensions installed you should be able to start a new AVRISP-MKII Clone example project from `File->New->Example Project`. You can then edit the code to suit your hardware, and compile it into a HEX file.

To program the Arduino Leonrdo board, you should use the `avrdude` tool. You can download and install that standalone, or you can install the Arduino IDE as that includes avrdude (although you may have to dig in the Arduino installation directory to find it).

Ivanov09, 2016-08-05:
sorry for my english.
Hi i'm working with AS7 and AS6.2 and AVRISP-MKII Clone does not work. :(
i created a new proyect form ASF with LUFA and i made the changes that you told, but nothing happends.

[image]
[image]
[image]

can you help me please!, 
[LUFA_PROJECTS_AVRISPMKII_CLONE1.zip](https://github.com/abcminiuser/lufa/files/404241/LUFA_PROJECTS_AVRISPMKII_CLONE1.zip)

i'm working with At90USB162, 16MHz, and USBTINYMKII, programmig  PDI to ATXMEGA32A4U

i hope you can helping me.

abcminiuser, 2016-08-07:
@Ivanov09 that indicates that the programmer is interfacing with Atmel Studio just fine, but it's failing to communicate with the target AVR device. What connections have you got between your clone programmer and the AVR?

Ivanov09, 2016-08-07:
thanks a lot Dean, the failure was a cold welding or bad weld, sorry for my english. Now it's working well.

abcminiuser, 2016-08-07:
Avoiding that is literally the point of that helper function - the PlexAPI library returns a custom type that compares equal to `None`, but `is None` evaluates to `False`.

MNdiaz, 2016-08-08:
Hi Dean, Thanks for Reply, good day, I continued doing examples but still the same error, proves more with which you told me, and I will continue investigating to learn more about the USB communication, I comment if I get better results.
I will try with the PlexAPI library, we are in contact, thanks for everything, good day.

abcminiuser, 2016-08-08:
Ignore the PlexAPI comment (since deleted) - my phone client posted that to the wrong issue (it was supposed to go on the Home-Assistant project).