abcminiuser/lufa #65: Wrong type for the wIndex parameter of the CALLBACK_USB_GetDescriptor function

matlo, 2015-11-27:
It should be uint16_t, not uint8_t.

abcminiuser, 2015-11-27:
Hi Matlo,

Yes, good catch. The `wIndex` field of the GetDescriptor standard request is actually only used in one circumstance; to retrieve string descriptors in a specific language (as it's possible for a device to have multiple localized versions of the same strings). Language IDs are 16-bit, so currently it's being truncated and isn't useful.

That said - the fact that no-one has *ever* complained in some 7 or 8 years makes me think that localized strings are not very popular. That said, there's no point having a non-working parameter around, so I'll fix this for the next release.

matlo, 2015-11-30:
Hi Dean,
Good guess, I had the issue with strings. I'm working on a USB proxy project ([matlo/serialusb](http://www.github.com/matlo/usbserial)), with a PC as a host, and an atmega32u4 as a device. The atmega32u4 is connected to the PC using a USB to UART adapter. I'm sending all the descriptors over serial into a RAM buffer, as well as an index (offset, wValue, wIndex, wLength).
Thanks for your work on LUFA!

abcminiuser, 2015-12-02:
Fixed in https://github.com/abcminiuser/lufa/commit/92e9cb71e290ea7e05d15f4761b8a07d786a8aa9 - let me know you have any other issues. Enjoy!

closed 2015-12-02 by abcminiuser