abcminiuser/lufa #66: Typos in USB_Device_SetConfiguration

matlo, 2015-12-08:
https://github.com/abcminiuser/lufa/blob/1d2680b3582107fc14e6d9db1204670fa52272bd/LUFA/Drivers/USB/Core/DeviceStandardReq.c#L155
I think there are two typos: 
* USE_SRAM_DESCRIPTORS should be USE_RAM_DESCRIPTORS
* MEMSPACE_SRAM should be MEMSPACE_RAM

abcminiuser, 2015-12-09:
SRAM means Static Random Access Memory, which is a variant of generic RAM/Random Access Memory. I chose that naming because that's what Atmel specifically call their internal RAM in the device data sheets, and wanted to keep the terminology the same so as not to confuse people.

I could change it now, however as all three targets (UC3, XMEGA and AVR8) use SRAM in their data sheets I think it's best left the way it is so as not to introduce a migration issue for no real benefit.

matlo, 2015-12-09:
Isn't the token named USE_RAM_DESCRIPTORS in LUFAConfig.h?
And the enum value MEMSPACE_RAM?

abcminiuser, 2015-12-09:
Nuts, you're right. I'll change the code but add a compatibility fix to accept the current naming as well.

abcminiuser, 2015-12-15:
Thanks!

Fixed in https://github.com/abcminiuser/lufa/commit/40e1caa2da06d6d4699d5dc3dafb3d34b81f23f7.   Seems you're really exercising the corner parts of LUFA! It's an issue only for applications that do not define `FIXED_NUM_CONFIGURATION` (something that is rare) but do define `USE_RAM_DESCRIPTORS`.

closed 2015-12-15 by abcminiuser

matlo, 2015-12-15:
Hi Dean,
You're right, the USB proxy I'm developing only has RAM descriptors, and the number of configurations depends on the target device.
Thanks!