abcminiuser/lufa #69: Are you planning support for the Teensy 3.X?

jayeye, 2016-01-17:
before I go ahead and start working on it, do you already have planned support for it? Can you think of any reason why it wouldn't work?

Thanks

abcminiuser, 2016-02-07:
Currently no - it's a wildly different architecture to the USB AVR8 devices used in the Teensy 1 and 2 (and which LUFA supports). While it would be possible to port the LUFA API to those devices, it would likely give terrible performance due to the compromises I had to make in order to make a compact library for the very underpowered AVR8 parts.

Using the native USB libraries for the larger parts is recommended, as it will most likely give the best performance with the least amount of work.

closed 2016-02-07 by abcminiuser

eltang, 2017-04-27:
@abcminiuser I saw you mention plans to convert LUFA to interrupt-driven operation a few times here and there. That would be a stepping stone to eventual Teensy 3.x support, correct?