abcminiuser/lufa #70: Demo and application INF files need to be signed

abcminiuser, 2016-02-07:
With Windows's gestapo driver signing now being truly enforced, we need signed versions of the LUFA demo INF driver files to be able to continue to use them on Windows machines. Apparently CDC-ACM now works on Windows 10 without any driver files (yay!) but Windows 8 and below still have a lot of market share, and RNDIS and other projects still require a signed driver even on Win10.

The list of driver INF files requiring signing are:
 - ./Bootloaders/CDC/LUFA CDC Bootloader.inf
 - ./Demos/Device/ClassDriver/DualVirtualSerial/LUFA DualVirtualSerial.inf
 - ./Demos/Device/ClassDriver/RNDISEthernet/LUFA RNDIS.inf
 - ./Demos/Device/ClassDriver/VirtualSerial/LUFA VirtualSerial.inf
 - ./Demos/Device/ClassDriver/VirtualSerialMassStorage/LUFA VirtualSerialMassStorage.inf
 - ./Demos/Device/ClassDriver/VirtualSerialMouse/LUFA VirtualSerialMouse.inf
 - ./Demos/Device/LowLevel/BulkVendor/WindowsDriver/LUFA_Bulk_Vendor_Demo.inf
 - ./Demos/Device/LowLevel/DualVirtualSerial/LUFA DualVirtualSerial.inf
 - ./Demos/Device/LowLevel/RNDISEthernet/LUFA RNDIS.inf
 - ./Demos/Device/LowLevel/VirtualSerial/LUFA VirtualSerial.inf
 - ./Projects/Benito/LUFA Benito Programmer.inf
 - ./Projects/LEDNotifier/LUFA LED Notifier.inf
 - ./Projects/SerialToLCD/LUFA SerialToLCD.inf
 - ./Projects/USBtoSerial/LUFA USBtoSerial.inf
 - ./Projects/Webserver/LUFA Webserver RNDIS.inf
 - ./Projects/XPLAINBridge/LUFA XPLAIN Bridge.inf

The vast majority of these are effectively identical, except for different VID/PID and descriptions. This means signing should be relatively easy once we get the base classes (CDC, RNDIS) figured out.

@ehajo @mitchjs will you be able to assist? Does someone else with a valid signing certificate wish to help out?

ehajo, 2016-02-07:
Hi dean,Seems your mailclient is eating my mail :)I already signed cdc and rndis.I will sign the other drivers too and send you a link.-------- Originalnachricht --------Betreff: [lufa] Demo and application INF files need to be signed (#70)Von: Dean Camera An: abcminiuser/lufa Cc: Hannes Jochriem With Windows's gestapo driver signing now being truly enforced, we need signed versions of the LUFA demo INF driver files to be able to continue to use them on Windows machines. Apparently CDC-ACM now works on Windows 10 without any driver files (yay!) but Windows 8 and below still have a lot of market share, and RNDIS and other projects still require a signed driver even on Win10.

The list of driver INF files requiring signing are:

./Bootloaders/CDC/LUFA CDC Bootloader.inf
./Demos/Device/ClassDriver/DualVirtualSerial/LUFA DualVirtualSerial.inf
./Demos/Device/ClassDriver/RNDISEthernet/LUFA RNDIS.inf
./Demos/Device/ClassDriver/VirtualSerial/LUFA VirtualSerial.inf
./Demos/Device/ClassDriver/VirtualSerialMassStorage/LUFA VirtualSerialMassStorage.inf
./Demos/Device/ClassDriver/VirtualSerialMouse/LUFA VirtualSerialMouse.inf
./Demos/Device/LowLevel/BulkVendor/WindowsDriver/LUFA_Bulk_Vendor_Demo.inf
./Demos/Device/LowLevel/DualVirtualSerial/LUFA DualVirtualSerial.inf
./Demos/Device/LowLevel/RNDISEthernet/LUFA RNDIS.inf
./Demos/Device/LowLevel/VirtualSerial/LUFA VirtualSerial.inf
./Projects/Benito/LUFA Benito Programmer.inf
./Projects/LEDNotifier/LUFA LED Notifier.inf
./Projects/SerialToLCD/LUFA SerialToLCD.inf
./Projects/USBtoSerial/LUFA USBtoSerial.inf
./Projects/Webserver/LUFA Webserver RNDIS.inf
./Projects/XPLAINBridge/LUFA XPLAIN Bridge.inf

The vast majority of these are effectively identical, except for different VID/PID and descriptions. This means signing should be relatively easy once we get the base classes (CDC, RNDIS) figured out.

@ehajo @mitchjs will you be able to assist? Does someone else with a valid signing certificate wish to help out?

ehajo, 2016-02-07:
Hi Dean,

here is the download to all signed drivers: http://www.ehajo.de/LUFA_signed_drivers.zip
Let me know if I missed one...

CDC-ACM is working without drivers on WIN10, yes, but you get a small "!" on the symbol and you do not get the right name of the device, its just "generic serial device" or something like that. With the driver you get the right name shown in the device-manager.