abcminiuser/lufa #72: [HID] No error on too big bootloader size

NicoHood, 2016-02-14:
If i try to compile the CDC bootloader with 2kb setting (for 32u4) I get a compile error.
For the HID I dont get a compile error.

It would be nice to a) get an error and b) maybe improve the error, so one know whats going on.

abcminiuser, 2016-02-15:
Good idea, added to my backlog. Unfortunately this won't have a nice answer I suspect (given that different parts have different maximum bootloader size restrictions) so it will probably end up being string parsing and a lookup table. Yuck.

NicoHood, 2016-02-15:
the bootloader size is in the makefile given. you need to enter it there. If the user adds a wrong value, then it is out of scope. but while developing you should be able to set a limit for yourself and see if you reached it already. Also it is confusing because lufa uses those jump tables. and therefor 96 bytes are missing and calculating it yourself might not be obvious.