abcminiuser/lufa #75: HID Bootloader Python Host APP errors.

NicoHood, 2016-03-15:
1. You should add `#!/usr/bin/python` as first line of the python file.
2. The intelhex link is down. `http://bialix.com/intelhex/` I used this instead: https://pypi.python.org/pypi/IntelHex
3. pywinusb is not linux compatible. I cannot run to python script :/

NicoHood, 2016-03-15:
I tried the c program which seems to work now. However I got serious bootloader problems now:

The sketch will not load. It seems to be a magic key problem but I noticed several weird errors. It must be something simple i guess? I even tried a fixed bootkey address which the arduino team uses:
```cpp
//uint16_t MagicBootKey ATTR_NO_INIT;

volatile uint8_t *const MagicBootKeyPtr = (volatile uint8_t *)0x8000;
#define MagicBootKey *MagicBootKeyPtr

/** Special startup routine to check if the bootloader was started via a watchdog reset, and if the magic application
 *  start key has been loaded into \ref MagicBootKey. If the bootloader started via the watchdog and the key is valid,
 *  this will force the user application to start via a software jump.
 */
void Application_Jump_Check(void)
{
	LEDs_Init();

	LEDs_SetAllLEDs(0);

	if(MCUSR & (1 << WDRF))
	LEDs_TurnOnLEDs(LEDS_LED1); //rx

	if(MagicBootKey == MAGIC_BOOT_KEY)
	LEDs_TurnOnLEDs(LEDS_LED2); //tx

	/* If the reset source was the bootloader and the key is correct, clear it and jump to the application */
	if ((MCUSR & (1 << WDRF)) && (MagicBootKey == MAGIC_BOOT_KEY))
	{
		MagicBootKey = 0;

		// cppcheck-suppress constStatement
		((void (*)(void))0x0000)();
	}
}
```

But only the first led goes on.
```makefile
MCU          = atmega32u4
ARCH         = AVR8
BOARD        = LEONARDO
F_CPU        = 16000000
F_USB        = $(F_CPU)
OPTIMIZATION = s
TARGET       = BootloaderHID
SRC          = $(TARGET).c Descriptors.c $(LUFA_SRC_USB)
LUFA_PATH    = ../../LUFA
CC_FLAGS     = -DUSE_LUFA_CONFIG_HEADER -IConfig/
LD_FLAGS     = -Wl,--section-start=.text=$(BOOT_START_OFFSET)

# Flash size and bootloader section sizes of the target, in KB. These must
# match the target's total FLASH size and the bootloader size set in the
# device's fuses.
FLASH_SIZE_KB        := 32
BOOT_SECTION_SIZE_KB := 4

avrdude:
	avrdude -patmega32u4 -cstk500v1 -P/dev/ttyACM0 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xCB:m -Uhfuse:w:0xD8:m -Ulfuse:w:0xFF:m
	avrdude -patmega32u4 -cstk500v1 -P/dev/ttyACM0 -b19200 -Uflash:w:./$(TARGET).hex:i -Ulock:w:0x2F:m
```

abcminiuser, 2016-03-21:
Good points @NicoHood!

For point #1: yes, I wasn't thinking much about being cross-platform, as evidenced by point 3 :). Looks like https://github.com/trezor/cython-hidapi is a decent cross-platform HID library, so I'll have to have a look at porting over the utility to make it properly cross platform.

NicoHood, 2016-03-21:
I used the c program for now, it works great. Even though I suggest using hidapi for linux instead of libusb.

abcminiuser, 2018-01-07:
Fixed in bf630b3add2ed197ec4c9f94f06d4dec4281d09d.

closed 2018-01-07 by abcminiuser