abcminiuser/lufa #90: Documentation cannot be generated

eltang, 2016-11-20:
I get the following error when I try to follow the instructions for using Doxygen to generate the documentation.

```
Executing "make doxygen" on all LUFA library elements.

/Library/Developer/CommandLineTools/usr/bin/make -C LUFA doxygen
grep: /Version.h: No such file or directory
 [DOXYGEN] : Configuration file "doxyfile" with parameters "QUIET=YES PROJECT_NUMBER="
if ( ( cat doxyfile ; echo "QUIET=YES" ; echo "PROJECT_NUMBER=" ) | doxygen - 2>&1 | grep -v "warning: ignoring unsupported tag" ;); then exit 1; fi;
warning: Tag `CLANG_ASSISTED_PARSING' at line 1010 of file `-' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag `CLANG_OPTIONS' at line 1018 of file `-' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
make[1]: *** [doxygen] Error 1
make: *** [doxygen] Error 2
```

abcminiuser, 2016-11-20:
Looks like you're using a version of Doxygen that is either too old to support that config option, or has been specifically built without it.

The good news is that the `CLANG_ASSISTED_PARSING` option isn't actually required and is set to `NO` in the config file, so you should be able to fix it by [removing this line](https://github.com/abcminiuser/lufa/blob/master/LUFA/doxyfile#L1010) from your local checkout before re-running `make doxygen`.

eltang, 2016-11-20:
I installed Doxygen using Homebrew, so it should be a fairly recent version, right?

I've gotten the documentation generated by doing a project-wide find-and-replace operation to add a `#` in front of every occurrence of `CLANG_ASSISTED_PARSING` and `CLANG_OPTIONS`. In the process, I was presented with the following lines many times.

```
 [DOXYGEN] : Configuration file "doxyfile" with parameters "QUIET=YES"
if ( ( cat doxyfile ; echo "QUIET=YES" ) | doxygen - 2>&1 | grep -v "warning: ignoring unsupported tag" ;); then exit 1; fi;
```

abcminiuser, 2016-11-20:
> I installed Doxygen using Homebrew, so it should be a fairly recent version, right?

Oh, a Mac - no idea, but IIRC Homebrew usually builds from source. I guess if that's the case they've probably configured it to build without the additional clang dependency.

Disabling the option by stripping it out should be fine, although I do wish Doxygen added an option to just ignore those sorts of unsupported or unknown settings if they aren't enabled.

eltang, 2016-11-20:
So can I close this issue then? By the way, did you receive my email about using NicoHood/HID?

abcminiuser, 2016-11-20:
Yes, I'll close it.

Got your email - will take a look as soon as I can (so much to do these days, so little time 😞).

closed 2016-11-20 by abcminiuser

ryanalexmartin, 2022-07-17:
I had a similar error and I was able to force building of the documentation with `make doxygen -i` to ignore errors.

Apteryks, 2023-09-17:
Hello,

I have this issues where make returns uncleanly from `make doxygen`, as seen in the output below.  I've used the latest lufa commit, (15cf7809c7d9e7a6e42f256309dd965b6c8cc20b), as well as the latest stable doxygen release (1.9.8), but the problem persists.  Perhaps this issue should be reopened?

```
starting phase `build-documentation'
Executing "make doxygen" on all LUFA library elements.

make -C LUFA doxygen
make[1]: Entering directory '/tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA'
 [DOXYGEN] : Configuration file "doxyfile" with parameters "QUIET=YES PROJECT_NUMBER=210130"
if ( ( cat doxyfile ; echo "QUIET=YES" ; echo "PROJECT_NUMBER=210130" ) | doxygen - 2>&1 | grep -v "warning: ignoring unsupported tag" ;); then exit 1; fi;
warning: Tag 'CLANG_ASSISTED_PARSING' at line 1129 of file '-' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_ADD_INC_PATHS' at line 1137 of file '-' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_OPTIONS' at line 1145 of file '-' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_DATABASE_PATH' at line 1158 of file '-' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'HTML_TIMESTAMP' at line 1305 of file '-' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'FORMULA_TRANSPARENT' at line 1626 of file '-' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'LATEX_TIMESTAMP' at line 1970 of file '-' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLASS_DIAGRAMS' at line 2320 of file '-' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_FONTNAME' at line 2362 of file '-' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_FONTSIZE' at line 2369 of file '-' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_TRANSPARENT' at line 2615 of file '-' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
/tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Common.h:306: warning: no matching file member found for 
static uint_reg_t GetGlobalInterruptMask(void)
Possible candidates:
 'static uint_reg_t GetGlobalInterruptMask(void) ATTR_ALWAYS_INLINE ATTR_WARN_UNUSED_RESULT' at line 306 of file /tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Common.h
/tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Common.h:328: warning: no matching file member found for 
static void SetGlobalInterruptMask(MACHINE_REG_t GlobalIntState)
Possible candidates:
 'static void SetGlobalInterruptMask(const uint_reg_t GlobalIntState) ATTR_ALWAYS_INLINE' at line 328 of file /tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Common.h
/tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Common.h:351: warning: no matching file member found for 
static void GlobalInterruptEnable(void)
Possible candidates:
 'static void GlobalInterruptEnable(void) ATTR_ALWAYS_INLINE' at line 351 of file /tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Common.h
/tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Common.h:371: warning: no matching file member found for 
static void GlobalInterruptDisable(void)
Possible candidates:
 'static void GlobalInterruptDisable(void) ATTR_ALWAYS_INLINE' at line 371 of file /tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Common.h
/tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Endianness.h:401: warning: no matching file member found for 
static uint16_t SwapEndian_16(uint16_t Word)
Possible candidates:
 'static uint16_t SwapEndian_16(const uint16_t Word) ATTR_WARN_UNUSED_RESULT ATTR_CONST ATTR_ALWAYS_INLINE' at line 401 of file /tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Endianness.h
/tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Endianness.h:432: warning: no matching file member found for 
static uint32_t SwapEndian_32(uint32_t DWord)
Possible candidates:
 'static uint32_t SwapEndian_32(const uint32_t DWord) ATTR_WARN_UNUSED_RESULT ATTR_CONST ATTR_ALWAYS_INLINE' at line 432 of file /tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA/Common/Endianness.h
[... cut ...]
make[1]: *** [Build/DMBS/DMBS/doxygen.mk:48: doxygen] Error 1
make[1]: Leaving directory '/tmp/guix-build-lufa-210130-1.15cf780.drv-0/source/LUFA'
make: *** [makefile:20: doxygen] Error 2
error: in phase 'build-documentation': uncaught exception:
%exception #<&invoke-error program: "make" arguments: ("-j" "4" "doxygen" "SHELL=/gnu/store/rib9g2ig1xf3kclyl076w28parmncg4k-bash-minimal-5.1.16/bin/sh") exit-status: 2 term-signal: #f stop-signal: #f> 
phase `build-documentation' failed after 2.1 seconds
```

Thanks.

abcminiuser, 2023-09-19:
Should be resolved now by 0200c233f30774935822ddb2b963a7ae0675edd9.