abcminiuser/lufa #92: LUFA USB module in Atmel Studio 7 appears to be out of date

Yorkshire-Jack, 2017-01-19:
I noticed this while moving a program written elsewhere over to Atmel Studio.  When I tried to use the ASF version of LUFA, there was a mismatch in the definition of CALLBACK_USB_GetDescriptor between my descriptors.c and the version of device.h brought into the project by ASF.

The relevant change  is commit https://github.com/abcminiuser/lufa/commit/92e9cb71e290ea7e05d15f4761b8a07d786a8aa9. In the version of LUFA downloaded from fourwalledcubicle `wIndex` is a `uint16`, whereas the version in AS is a `uint8` - as it was before the commit.

This seems to appear both when adding the LUFA USB module through ASF wizard, and when opening the CLASS CDC demo project included in the LUFA extension. The easiest way to reproduce it is to open the CDC demo and look at the definition in descriptors.h and devices.h

I have not looked to see if anything else is out of date.

abcminiuser, 2017-01-22:
Ah yes, I need to make a new build for the gallery, I'll add it to my TODO list for the week.

Yorkshire-Jack, 2017-02-15:
I have just realised that in my original post above I thought I was comparing the ASF extension to the fourwalledcubicle download.  But the code I was migrating into Atmel Studio actually used a newer version from Git.  Both the ASF and fourwalledcubicle versions are actually the same, version 151115.

So this shouldn't be a bug report that the ASF version is behind, but rather a feature request for a new release. Sorry for the noise.