abcminiuser/lufa #95: Duplicate ISR() under some condition

osamuaoki, 2017-02-02:
For following code:

- lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c
- lufa/LUFA/Drivers/USB/Core/UC3/USBInterrupt_UC3.c

I see duplicate definition of ISR(...).  Let me focus on AVR side for simplicity.

For ISR definition on line 77, it is always there.
For ISR definition on line 261, it is conditional with:

#if defined(INTERRUPT_CONTROL_ENDPOINT) && defined(USB_CAN_BE_DEVICE)

Do you expect to duplicate definition if INTERRUPT_CONTROL_ENDPOINT and USB_CAN_BE_DEVICE are defined?  

Is this some fancy function overloading?  But this is C code ...

eltang, 2017-02-02:
Just yesterday I tried compiling my project with INTERRUPT_CONTROL_ENDPOINT defined, and the resulting binary did not work.

abcminiuser, 2017-02-03:
There should not be a duplicate definition; one is `USB_GEN_vect`, while the other is `USB_COM_vect`. These are different vectors in the device, and the handlers will be called under different circumstances.

- USB_GEN is the USB General Interrupt Vector, called when the device/host is suspended, reset, VBUS changes, etc.

- USB_COM is the USB Common Interrupt Vector, called when an endpoint or pipe whose interrupts are enabled fires

osamuaoki, 2017-02-03:
Hi,

On Fri, Feb 03, 2017 at 12:30:46AM -0800, Dean Camera wrote:
> There should not be a duplicate definition; one is USB_GEN_vect, while the
> other is USB_COM_vect. These are different vectors in the device, and the
> handlers will be called under different [...]

I am just a first time reader of this source and I did not parse this
part.  No wonder.  I hope this report helped you to improve this code
base.

By the way, as for LUFA HID bootloader, according to the TMK firmware
source, LUFA is rather big:

 * Size of Bootloaders in bytes:
 *   Atmel DFU loader(ATmega32U4)   4096
 *   Atmel DFU loader(AT90USB128)   8192
 *   LUFA bootloader(ATmega32U4)    4096
 *   Arduino Caterina(ATmega32U4)   4096
 *   USBaspLoader(ATmega***)        2048
 *   Teensy   halfKay(ATmega32U4)   512
 *   Teensy++ halfKay(AT90USB128)   1024

But nanoBoot written in GAS is 512 byte and in MIT license.
  https://github.com/volium/nanoBoot.git

You may consider this as one optional component.

Osamu

eltang, 2017-02-04:
It turns out that my problem was unrelated. It seems that HID_Device_USBTask() cannot be called from an interrupt when INTERRUPT_CONTROL_ENDPOINT is defined.

@abcminiuser Where would be the right place for me to add a note about this?

closed 2017-10-01 by abcminiuser