abcminiuser/lufa #96: Request for MSC-CDC Composite Bootloader for ATmega32u4

SandeepanSengupta, 2017-03-28:
Is it possible to make a MSC-CDC Composite Bootloader for ATmega32u4 that can switch in between MSC and CDC mode on certain event (like rapidly pressing reset button twice)

@watterott have a similer implementation based on ATSAMD21
[https://github.com/watterott/SAM-BAR](url)

abcminiuser, 2017-03-30:
It's possible, but it wouldn't fit into the bootloader space (the MS class one already takes up 6KB, so has to spill out of the bootloader region). While technically feasible, the limited flash space means you're be better off just loading in the bootloader class you prefer.

SandeepanSengupta, 2017-03-31:
@abcminiuser 
Sir, MSC-CDC composite bootloader will be a pretty useful (and cool) thing to have as a part of my university project but unfortunately I lack the technical skill to do so. Any aid will be a bliss.
Regards