abcminiuser/lufa #97: hid_bootloader_cli is based on rather older teensy_loader_cli

osamuaoki, 2017-04-02:
It seems teensy_loader_cli which your hid_bootloader_cli is forked from has been updated to accommodate newer arms with larger buffer allocation etc.

I updated newer teensy_loader_cli  to accept LUFA firmware.  (Unlike your current approach to try for both, I made USB ID choice to be deterministic.  If compiled code is called as hid_bootloader_cli, it only works with LUFA firmware.

See it in the branched teensy_loader_cli repo
* https://github.com/osamuaoki/teensy_loader_cli
I tried to merge as much fixes as you did into htis too.

Also code itself is
* https://github.com/osamuaoki/teensy_loader_cli/blob/master/teensy_loader_cli.c

I hope this patch set is more easy to track teensy code in future.

Osamu

maqifrnswa, 2017-05-12:
Thank you, this is helpful to me.
I get an error using LUFA's HID uploader on Marlin 3D Printer firmware:
```
./hid_bootloader_cli -mmcu=at90usb1286 -v /tmp/arduino_build_378673/Marlin.ino.hex
Teensy Loader, Command Line, Version 2.0
error reading intel hex file "/tmp/arduino_build_378673/Marlin.ino.hex"
```

that I don't get with teensy's newest version (or osamuaoki's version)
```
./teensy_loader_cli -mmcu=at90usb1286 -v /tmp/arduino_build_378673/Marlin.ino.hex
USB Vendor ID=0X16C0, Product ID=0X0478 **(THIS LINE IS IN OSAMUAOKI'S FORK)**
Teensy Loader, Command Line, Version 2.1
Read "/tmp/arduino_build_378673/Marlin.ino.hex": 70002 bytes, 53.8% usage
Unable to open device
```

tenten8401, 2018-04-17:
I'd love to see this one fixed if possible since it's basically preventing me from flashing any .hex file I try with the HID bootloader on my Printrboard F5. I posted a [$5 bounty](https://www.bountysource.com/issues/43710775-hid_bootloader_cli-is-based-on-rather-older-teensy_loader_cli) on bountysource if anyone's interested in trying to work on this (or at least fix the error reading issue).

As maqifrnswa stated though, I don't see the issue on the newest teensy version of the loader.

Here's the hex file I'm trying to upload on my Printrboard: [firmware.zip](https://github.com/abcminiuser/lufa/files/1918020/firmware.zip)

If I'm just being dumb and doing something wrong here then please let me know.
```
./hid_bootloader_cli -mmcu=at90usb1286 /home/isaaca/Marlin/.pioenvs/printrboard_revf/firmware.hex
Teensy Loader, Command Line, Version 2.0
error reading intel hex file "/home/isaaca/Marlin/.pioenvs/printrboard_revf/firmware.hex"```

tenten8401, 2018-04-17:
Alright, well I took the new Teensy Loader 2.1 and made a quick fix for it using the following from the current loader:
```
if (!libusb_teensy_handle)
	libusb_teensy_handle = open_usb_device(0x03eb, 0x2067);
```

I was able to flash my firmware successfully using it. Here's a gist of my changes: https://gist.github.com/tenten8401/9c575d95c483f64dbf9a378fc0cf9cfb

I'm sure it'll come in useful for someone, unless the main version gets some love.

abcminiuser, 2021-08-15:
PJRC code has been rebased; closing for now until someone confirms this is still relevant with the latest HEAD.

closed 2021-08-15 by abcminiuser