abcminiuser/lufa #98: LUFA fails to wake up host

eltang, 2017-05-11:
The following code seems to have no effect when run on an ATmega32u4 whose host is a Mac.
```c
if (USB_DeviceState == DEVICE_STATE_Suspended)
    if (USB_Device_RemoteWakeupEnabled)
        USB_Device_SendRemoteWakeup();
```

abcminiuser, 2017-05-11:
What is the state of `USB_Device_RemoteWakeupEnabled`?

Make sure you have enabled `USB_CONFIG_ATTR_REMOTEWAKEUP` in your configuration descriptor's `ConfigAttributes` element, along with the mandatory `USB_CONFIG_ATTR_RESERVED` attribute and (optional) `USB_CONFIG_ATTR_SELFPOWERED` attribute. Without this option, the host won't enable remote wakeup and you won't be able to trigger a wake from suspended mode.

eltang, 2017-05-12:
They have respective values of `1` and `USB_CONFIG_ATTR_RESERVED | USB_CONFIG_ATTR_REMOTEWAKEUP`, just as expected. In the meantime, I've also discovered that the ATmega32U4 renders the mouse unable to wake up the computer as well.

eltang, 2017-05-12:
I've found something even stranger. It turns out that the ATmega32U4 is able to wake up the computer, but only if it's been suspended for a long period of time.

eltang, 2017-05-13:
Whoops, after some more rigorous testing I discovered that the problem is with my code and not with LUFA.

closed 2017-05-13 by eltang