abcminiuser/lufa #99: multiple definition of `__vector_17'

LightningStalker, 2017-07-26:
Hey guys
I try to compile LUFA and get this weird error.

`/home/robert/src/lufa/Projects/AVRISP-MKII/Lib/ISP/ISPTarget.c:119: multiple definition of `__vector_17'
obj/SoftUART.o:/home/robert/src/lufa/Projects/XPLAINBridge/Lib/SoftUART.c:102: first defined here
/usr/lib/gcc/avr/4.8.1/../../../avr/bin/ld: Disabling relaxation: it will not work with multiple definitions
collect2: error: ld returned 1 exit status
../../LUFA/Build/DMBS/DMBS/gcc.mk:241: recipe for target 'XPLAINBridge.elf' failed
make[2]: *** [XPLAINBridge.elf] Error 1
make[2]: Leaving directory '/home/robert/src/lufa/Projects/XPLAINBridge'
makefile:44: recipe for target 'XPLAINBridge/' failed
make[1]: *** [XPLAINBridge/] Error 2
make[1]: Leaving directory '/home/robert/src/lufa/Projects'
makefile:19: recipe for target 'all' failed
make: *** [all] Error 2`

Any ideas?

LightningStalker, 2017-08-03:
OS is Zorin OS (Ubuntu based)

LightningStalker, 2017-08-14:
Also happening on Linux Mint 18.2
It doesn't matter weather it's the latest commit or 170418

abcminiuser, 2017-08-23:
Whoops, all fixed in 01e75290a0ea730e69c8465a17626c6458d19a9b. The XPLAINBridge project uses some sources from the AVRISP project, but I was compiling in the ISP programming protocol code which isn't needed, and which uses an interrupt vector internally that the XPLAINBridge project uses for other purposes.

Since the XPLAINBridge project doesn't use the ISP protocol compiling in the module isn't needed, and only results in the linker conflicts you're seeing there from the duplicate interrupt handler definition.

closed 2017-08-23 by abcminiuser

LightningStalker, 2017-08-23:
Thanks will checkout

LightningStalker, 2017-08-24:
Ok, finally got around to doing a test compile, and apparently XPLAINBridge is apparently still looking for the ISP protocol for some reason.  Here's what happens:
`../AVRISP-MKII/obj/V2Protocol.o: In function `V2Protocol_Init':
V2Protocol.c:(.text.V2Protocol_Init+0xe): undefined reference to `ISPTarget_ConfigureRescueClock'
../AVRISP-MKII/obj/V2Protocol.o: In function `V2Protocol_ProcessCommand':
V2Protocol.c:(.text.V2Protocol_ProcessCommand+0x8a): undefined reference to `ISPProtocol_EnterISPMode'
V2Protocol.c:(.text.V2Protocol_ProcessCommand+0x19e): undefined reference to `ISPProtocol_LeaveISPMode'
V2Protocol.c:(.text.V2Protocol_ProcessCommand+0x1a6): undefined reference to `ISPProtocol_ProgramMemory'
V2Protocol.c:(.text.V2Protocol_ProcessCommand+0x1ae): undefined reference to `ISPProtocol_ReadMemory'
V2Protocol.c:(.text.V2Protocol_ProcessCommand+0x1b4): undefined reference to `ISPProtocol_ChipErase'
V2Protocol.c:(.text.V2Protocol_ProcessCommand+0x1bc): undefined reference to `ISPProtocol_ReadFuseLockSigOSCCAL'
V2Protocol.c:(.text.V2Protocol_ProcessCommand+0x1c4): undefined reference to `ISPProtocol_WriteFuseLock'
V2Protocol.c:(.text.V2Protocol_ProcessCommand+0x1ca): undefined reference to `ISPProtocol_SPIMulti'
collect2: error: ld returned 1 exit status
../../LUFA/Build/DMBS/DMBS/gcc.mk:241: recipe for target 'XPLAINBridge.elf' failed
make[2]: *** [XPLAINBridge.elf] Error 1
make[2]: Leaving directory '/home/robert/src/lufa/Projects/XPLAINBridge'
makefile:44: recipe for target 'XPLAINBridge/' failed
make[1]: *** [XPLAINBridge/] Error 2
make[1]: Leaving directory '/home/robert/src/lufa/Projects'
makefile:19: recipe for target 'all' failed
make: *** [all] Error 2
`

abcminiuser, 2017-10-01:
Did you perform a `make clean` beforehand, and pull the latest code? Your link step still refers to `ISPProtocol.o` which should no longer be a prerequisite of the ELF build target.

szg0000, 2018-07-04:
Bug confirmed. Same problem here.