abcminiuser/lufa #102: Intermittent PDI failure: Failed to enter programming mode

mattairtech, 2017-09-24:
I had been receiving "Failed to enter programming mode. Error status received from tool: 0x03." from Atmel Studio when programming XMEGA chips using PDI (at least one customer reported the problem as well). This was more probable to occur when temperatures were higher (especially noticeable after I started programming chips shortly after oven reflow and the boards were still warm). Sometimes multiple tries would work.

Code 0x03 is XPRG_ERR_TIMEOUT. According to the datasheet, there is a maximum time of 100us allowed between bringing the PDI DATA line high, and sending the clock signal to PDI CLK (RESET). In the code, there is a required delay between these two events, and it used to be 1us. It was then changed at some point (at least two years ago) to exactly 100us. This is right at the edge. The internal RC oscillator (ULP) used by the XMEGA for internal timing purposes (like the 100us maximum), is very sensitive to temperature (+/- 30%), so this would explain the intermittent nature.

In XPROGTarget_EnableTargetPDI() line 55 (for PDI) and line 75 (for TPI) change:
```
_delay_us(100);
```
to
```
_delay_us(20);
```
This solved all problems for me and the customer.
The value of 20 may be too low, but it has been well tested. Paranoia from both sides.

abcminiuser, 2017-10-01:
Looking over appnote AVR1612, I think my code's subtly even wronger; I should be pulling the PDI CLOCK line high, and PDI DATA line low for 90ns to trigger an external reset, and *then* pulling PDI DATA high for 16 PDI clock cycles. Right now I'm skipping the first part, which is likely the real source of the issues.

Would you mind testing the below patched function? I've been digging through boxes trying to find a suitable programmer board this afternoon, with no luck yet...

```
void XPROGTarget_EnableTargetPDI(void)
{
	IsSending = false;

	/* Set Tx and XCK as outputs, Rx as input */
	DDRD |=  (1 << 5) | (1 << 3);
	DDRD &= ~(1 << 2);

	/* Set Tx (PDI CLOCK) high, DATA line low for at least 90ns to disable /RESET functionality */
	PORTD |=  (1 << 5);
	PORTD &= ~(1 << 3);
	_delay_us(100);

	/* Set DATA line high (enables PDI interface after 16 PDI CLK cycles) */
	PORTD |= (1 << 3);
	_delay_us(20);

	/* Set up the synchronous USART for XMEGA communications - 8 data bits, even parity, 2 stop bits */
	UBRR1  = ((F_CPU / 2 / XPROG_HARDWARE_SPEED) - 1);
	UCSR1B = (1 << TXEN1);
	UCSR1C = (1 << UMSEL10) | (1 << UPM11) | (1 << USBS1) | (1 << UCSZ11) | (1 << UCSZ10) | (1 << UCPOL1);

	/* Send two IDLEs of 12 bits each to enable PDI interface (need at least 16 idle bits) */
	XPROGTarget_SendIdle();
	XPROGTarget_SendIdle();
}```

mattairtech, 2017-10-03:
I think you are right. I don't have a board to test the above code right now, however, I do have an ATmega32u2 based AVRISPmkII programmer that uses similar code. I have three analog switches and a level shifter (with enable) onboard, so my code uses a different procedure to deal with these devices. I have been setting reset/pdi_clk low for 120us, then pdi_data low for 120us, then pdi_data high for now 20us. In your case, you probably don't have to set pdi_data low first (internal pulldown on MCU pin), but in my case, I had to set them separately because of the level shifter (though I cannot remember the details of exactly why, but it was needed for me). I have been using this procedure for years.

abcminiuser, 2017-12-04:
Found a suitable board and ordered the cables, they just arrived. Confirmed the new PDI entry sequence still works.

Fixed in https://github.com/abcminiuser/lufa/commit/173164ce426377fc31a1df30b1017c1fab52b96e - thanks!

closed 2017-12-04 by abcminiuser