abcminiuser/lufa #104: Questions about version 170418

matlo, 2017-10-18:
Hi,

I reviewed the changelog of this version and was wondering the impacts of the following issues:

> Fixed void pointer arithmetic in the Serial_SendData() functions for AVR8 and XMEGA architectures

This does not seem impacting. Is it just about making the code compliant with the C standard?

> Fixed invalid endpoint indexes causing memory corruption in device Clear/Set Feature standard requests (thanks to Peter Popovec)

This one got more of my attention. What impact does this have? Stack corruption? What was the impact if the selected endpoint was in the allowed range, but not present in the configuration?

Regards

abcminiuser, 2017-10-23:
> This does not seem impacting. Is it just about making the code compliant with the C standard?

Yes, it was relying on undefined pointer arthmetic on `void*`, which AVR-GCC will treat as a `uint8_t*` for arithmetic purposes. The fix makes the code compliant to the C standard and removes some compile warnings.

> This one got more of my attention. What impact does this have? Stack corruption? What was the impact if the selected endpoint was in the allowed range, but not present in the configuration?

The host could poke at endpoint descriptor tables beyond the configured maximum (up to the absolute endpoint maximum address of the XMEGA architecture), which could correct RAM. In general LUFA was not designed to be secure, it was designed to be small (and fast enough on the original architecture).

matlo, 2017-10-23:
Thanks for the answer! Any thought on the impact if the target endpoint is within the correct range, but not a configured endpoint?

abcminiuser, 2017-11-07:
If it's within the configured maximum endpoint allocations, not much would happen - the host would be able to fiddle with the endpoint STALL control bit, which does nothing on an endpoint that has not been configured.

matlo, 2017-11-07:
Thanks for the answer :)

closed 2017-11-07 by matlo