abcminiuser/lufa #107: USB_CONFIG_ATTR_SELFPOWERED for keyboard?

NicoHood, 2017-12-29:
https://github.com/abcminiuser/lufa/blob/master/Demos/Device/ClassDriver/Keyboard/Descriptors.c#L98
```c
	.Config =
		{
			.Header                 = {.Size = sizeof(USB_Descriptor_Configuration_Header_t), .Type = DTYPE_Configuration},

			.TotalConfigurationSize = sizeof(USB_Descriptor_Configuration_t),
			.TotalInterfaces        = 1,

			.ConfigurationNumber    = 1,
			.ConfigurationStrIndex  = NO_DESCRIPTOR,

			.ConfigAttributes       = (USB_CONFIG_ATTR_RESERVED | USB_CONFIG_ATTR_SELFPOWERED),

			.MaxPowerConsumption    = USB_CONFIG_POWER_MA(100)
},
```

Why is `USB_CONFIG_ATTR_SELFPOWERED` used for an USB keyboard? Wouldnt it make much more sense to remove it? I know this was possibly copied from other samples, but is there any noticeable difference in the behavior from the host site? Because normally i expect that microcontroller projects for usb are bus powered.

As a second question, wouldnt it rather make sense to enable remote wakeup here? In the corresponding [lufa config](https://github.com/abcminiuser/lufa/blob/master/Demos/Device/ClassDriver/Keyboard/Config/LUFAConfig.h#L78) remote wakeup is enabled/supported. Or do you only recommend to set this config if its actually used inside the code itself?

abcminiuser, 2017-12-31:
I just copy-pasted from the other examples. The `USB_CONFIG_ATTR_SELFPOWERED` tells the host the device has the *capability* to be self-powered (e.g. from an internal battery or external power supply). If this flag is set, the host can query via a `GetStatus` standard request to the device to determine if the device is currently self powered or being powered from the host.

If you set the `USB_CONFIG_ATTR_SELFPOWERED` flag, you can indicate to the host that you are currently self powered by setting `USB_Device_CurrentlySelfPowered` to `true`. The host will then take this into account when determining if it has sufficient current to power other USB devices freshly plugged into a hub port.

Indicating self powered capabilities but always returning a status of being host powered is fine, as it just means a little extra book-keeping on the host to manage the power consumption across the hub. If you definitely don't ever self-power, you can remove that flag from the demo.

Adding the remote wake flag is probably a good idea for the demos - I'll add it to my TODO list. I'm trying to get back into personal projects again, so I'm building a list of little and big things I want to do...

NicoHood, 2018-01-02:
Thanks for the detailed answer.

Now the question is, if the host even cares about this information, since (from what ive heard) some devices just ignore the usb specs in some cases, like this one. I am not sure if its worth for me to add a hardware detection if my PSU is plugged in or not, i will just remove it, to have less flash usage.

For the remote wakeup, we should track it in another issue: https://github.com/abcminiuser/lufa/issues/112

closed 2018-01-02 by NicoHood

abcminiuser, 2018-01-04:
Documentation updated in 0bf24677497dee28f092f882e1db396d907a3fc1.

NicoHood, 2018-01-04:
Perfect!