abcminiuser/lufa #127: CDC bootloader: incorrect check of BOOTRST fuse

bkell, 2018-04-01:
The CDC bootloader checks the BOOTRST fuse using this line:

`if (BootloaderAPI_ReadFuse(GET_HIGH_FUSE_BITS) & FUSE_BOOTRST)`

However FUSE_BOOTRST is an inverted mask (defined as ~_BV(0) for the device I'm using), meaning the conditional is true if any other high fuse bits are unprogrammed. Instead, a conditional like this should be used to check the fuse:

`if (!(BootloaderAPI_ReadFuse(GET_HIGH_FUSE_BITS) & ~FUSE_BOOTRST))`

bkell, 2018-04-01:
The printer, mass storage, and DFU bootloaders have this issue as well.

abcminiuser, 2018-04-02:
Oof - that was dumb, I always forget about the inverted fuse bits. Fixed in 1e9148db894bca208d1076c291e85e134a2f3308 - thanks!

closed 2018-04-02 by abcminiuser