abcminiuser/lufa #133: Command interpreter demo over USB Virtual Serial CDC

szg0000, 2018-07-10:
There are a bunch of demos in this project. I wanna see a command interpreter demo project on a microcontroller, which opens a virtual serial port over USB and can type&run simple programmer defined commands.

NicoHood, 2018-07-10:
This has nothing to do with lufa, and more a pc side problem you are facing. You can even do that with a non-usb arduino. Just write some python code that wraps serial to cmd. However this has some risks, depending on what you are doing. I guess you can close this issue...

szg0000, 2018-07-10:
> You can even do that with a non-usb arduino. 

I would like to use USB, thats why I post here, in an USB project... :)

> This has nothing to do with lufa, and more a pc side problem you are facing. 

What do you mean? I would like to connect to an USB Atmel microcontroller via USB CDC connection trough a virtual COM port. I already did it. From PC side a [putty.exe](https://www.putty.org/) was enough to connect.

I want more than just sending characters to a PC terminal. I want to have a command interpreter, and it must be impemented at the  **microcontroller** side.

NicoHood, 2018-07-11:
So you want to run commands on the microcontroller? Which commands? Setting gpios or what?

Even if someone is willing to follow your request, what should he implement? "commands" is a large topic. Which commands? What should they do?

If you want to run pc commands (like controlling your music player from the command line) you MUST implement a PC tool. How can the PC know what the microcontroller wants from it? And again: This is not related to lufa, not even the microcontroller code, as a normal usart + ftdi is also suitable for this. Or at least it makes no difference what MCU you use.

You request is out of place here. Sorry.

NicoHood, 2018-07-12:
this has nothing to do with lufa. You better ask in the Arduino Forums.

closed 2018-08-19 by abcminiuser