abcminiuser/lufa #138: How i can usw the avrisp-mkii protocol in atmega16u2

mcuOne, 2018-09-02:
Hello my friend. How can I fully use the mkii protocol on the atmega16u2? So also the TPI protocol. For testing purposes, I use an Arduino UNO.

NicoHood, 2018-09-02:
How about compiling the project?
https://github.com/abcminiuser/lufa/tree/master/Projects/AVRISP-MKII

I am not sure if its possible to fit into the small flash/ram of the 16u2.

mcuOne, 2018-09-02:
I do not think the RAM is too small. KosakaLab did manage it: https://make.kosakalab.com/make/electronic-work/avrisp-mk2/uno-r3_avrisp-mk2_en/ At first I thought KasakaLab uses Lufa Lib, but it uses TPI programming probably his own method. Or am I wrong?

abcminiuser, 2018-10-21:
It should work fine on the tiny 8KB parts, but you'll have to disable ISP programming and enable TPI programming only. In the `Config/AppConfig.h` header of the project comment out `ENABLE_ISP_PROTOCOL` and leave `ENABLE_XPROG_PROTOCOL` enabled.

closed 2018-12-04 by abcminiuser