abcminiuser/lufa #140: BootloaderPrinter for atmega32u4

alberiolima, 2018-10-29:
BootloaderPrinter
Bootloader very big to atmega32u4 
*****
avr-gcc (GCC) 5.4.0
... [SIZE]    : Determining size of "BootloaderPrinter.elf"
avr-size --mcu=at90usb1287 --format=avr BootloaderPrinter.elf
AVR Memory Usage
----------------
Device: at90usb1287
Program:    3986 bytes (3.0% Full)
(.text + .data + .bootloader)
Data:        248 bytes (3.0% Full)
(.data + .bss + .noinit)

****** CHANGING to
MCU          = atmega32u4
BOARD        = LEONARDO
F_CPU        = 16000000

FLASH_SIZE_KB         = 32
BOOT_SECTION_SIZE_KB  = 4

=====>>>>
[LNK]     : Linking object files into "BootloaderPrinter.elf"
avr-gcc obj/BootloaderPrinter.o obj/Descriptors.o obj/BootloaderAPI.o obj/BootloaderAPITable.o obj/HIDParser.o obj/Device_AVR8.o obj/EndpointStream_AVR8.o obj/Endpoint_AVR8.o obj/Host_AVR8.o obj/PipeStream_AVR8.o obj/Pipe_AVR8.o obj/USBController_AVR8.o obj/USBInterrupt_AVR8.o obj/ConfigDescriptors.o obj/DeviceStandardReq.o obj/Events.o obj/HostStandardReq.o obj/USBTask.o obj/AudioClassDevice.o obj/CCIDClassDevice.o obj/CDCClassDevice.o obj/HIDClassDevice.o obj/MassStorageClassDevice.o obj/MIDIClassDevice.o obj/PrinterClassDevice.o obj/RNDISClassDevice.o obj/AndroidAccessoryClassHost.o obj/AudioClassHost.o obj/CDCClassHost.o obj/HIDClassHost.o obj/MassStorageClassHost.o obj/MIDIClassHost.o obj/PrinterClassHost.o obj/RNDISClassHost.o obj/StillImageClassHost.o -o BootloaderPrinter.elf -lm -Wl,-Map=BootloaderPrinter.map,--cref -Wl,--gc-sections -Wl,--relax -mmcu=atmega32u4 -Wl,--section-start=.text=0x7000 -Wl,--section-start=.apitable_trampolines=0x7FA0 -Wl,--undefined=BootloaderAPI_Trampolines -Wl,--section-start=.apitable_jumptable=0x7FE0 -Wl,--undefined=BootloaderAPI_JumpTable -Wl,--section-start=.apitable_signatures=0x7FF8 -Wl,--undefined=BootloaderAPI_Signatures
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: section .apitable_trampolines loaded at [0000000000007fa0,0000000000007fb7] overlaps section .data loaded at [0000000000007f2a,0000000000008001]
collect2: error: ld returned 1 exit status
../../LUFA/Build/DMBS/DMBS/gcc.mk:249: recipe for target 'BootloaderPrinter.elf' failed
make: *** [BootloaderPrinter.elf] Error 1
*******

so I modified the code by removing "static void ParseIntelHEXByte (const char ReadCharacter)" and compiled

****
avr-gcc (GCC) 5.4.0
..
 [SIZE]    : Determining size of "BootloaderPrinter.elf"
avr-size --mcu=atmega32u4 --format=avr BootloaderPrinter.elf
AVR Memory Usage
----------------
Device: atmega32u4
Program:    3392 bytes (10.4% Full)
(.text + .data + .bootloader)
Data:        291 bytes (11.4% Full)
(.data + .bss + .noinit)
**

but "ParseIntelHEXByte" is the main function

thank you if you can help

abcminiuser, 2018-12-05:
Can you try the latest HEAD please? I've optimized the main bootloader code to only store 16-bit offsets rather than 32-bit when parsing out the HEX values, since the 32KB devices won't need the full 32-bits of addressing.

alberiolima, 2018-12-05:
I got the same result, I believe the problem is the size of the bootloader
[image]

alberiolima, 2018-12-05:
[image]

abcminiuser, 2018-12-05:
That's my fault, I pushed the fix (74dcaa91458cbe8c6b6cbbb246d87e379f32f0e0) to my internal sever, but not to GitHub. My internal CI shows it as now building successfully for the ATMEGA32U4 with a 4KB (2K word) bootloader size under the latest Ubuntu release's standard `avr-gcc` packages.

alberiolima, 2018-12-05:
I was amazed at what can be done with LUFA, I developed a device that captures the data simulating a printer and sends by tcp.

alberiolima, 2018-12-05:
Yes it works. solved.
Thank you very much!

closed 2019-10-27 by abcminiuser