abcminiuser/lufa #144: Add MIDI-CI Support

brandenbyers, 2019-01-24:
# Add MIDI Capability Inquiry (MIDI-CI) Support

Are there any plans to implement the MIDI-CI spec? If there aren't current plans, is LUFA open to it being added?

For reference, see the [MIDI-CI spec](https://www.midi.org/articles/midi-manufacturers-association-mma-adopts-midi-capability-inquiry-midi-ci-specification) (sorry I can't post a direct link; the spec is behind a free user signup).

brandenbyers, 2019-01-29:
If I’m the first one asking for this, then my guess is there aren’t too many people crying out for this in LUFA. 😀

There still isn’t much of any hardware adoption of MIDI-CI yet (the spec was released less than a year ago), but now that the roadmap for MIDI 2.0 was officially announced this month, there should be much more interest in this incremental step toward MIDI 2.0 with MIDI-CI.

I’m specifically interested in MIDI-CI for iOS development and bi-directional communication with hardware controllers. But I don’t have any hardware to test with. So I’m planning to build my own using QMK, which relies on LUFA for MIDI on AVR and LUFA + Chibios for ARM.

I have no idea if many hardware manufacturers are using LUFA for MIDI controllers, but if they are, this could be a little push they need to start implementing MIDI-CI. As a DIY MIDI controller maker, I’ve been using LUFA for midi for the past couple of years with great success. So whether you decide to implement this or not; thank you for all the work you have done!

I’ll continue digging into the spec too and see if I can contribute despite my lack of hardware programming experience.

Teknel, 2022-01-17:
Adding MIDI 2.0 support will be a very interesting approach for LUFA, because is true that is not very common until now, but opens a very powerful window to new devices.
We are working in a home control devices and platform based on MIDI 2.0.
Mainly in SysEx8 functions that let to transmit 8bits data.