abcminiuser/lufa #145: Make error

ReekyStive, 2019-01-30:
I installed gcc, g++ on my Ubuntu
Then I installed doxygen by
`sudo apt install doxygen`
And run
`make doxygen`
But I can't compile it

`reekystive@reeky:~/midi/lufa$ make doxygen
Executing "make doxygen" on all LUFA library elements.

make -C LUFA doxygen
make[1]: Entering directory '/home/reekystive/midi/lufa/LUFA'
 [DOXYGEN] : Configuration file "doxyfile" with parameters "QUIET=YES PROJECT_NUMBER=170418"
if ( ( cat doxyfile ; echo "QUIET=YES" ; echo "PROJECT_NUMBER=170418" ) | doxygen - 2>&1 | grep -v "warning: ignoring unsupported tag" ;); then exit 1; fi;
make[1]: Leaving directory '/home/reekystive/midi/lufa/LUFA'
make -C Demos doxygen
make[1]: Entering directory '/home/reekystive/midi/lufa/Demos'
make -C Device doxygen
make[2]: Entering directory '/home/reekystive/midi/lufa/Demos/Device'
make -C ClassDriver doxygen
make[3]: Entering directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver'
make[4]: Entering directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver/AudioInput'
/bin/sh: 1: avr-size: not found
/bin/sh: 1: avr-size: not found
 [DOXYGEN] : Configuration file "doxyfile" with parameters "QUIET=YES"
if ( ( cat doxyfile ; echo "QUIET=YES" ) | doxygen - 2>&1 | grep -v "warning: ignoring unsupported tag" ;); then exit 1; fi;
make[4]: Leaving directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver/AudioInput'
make[4]: Entering directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver/AudioOutput'
/bin/sh: 1: avr-size: not found
/bin/sh: 1: avr-size: not found`

Then I run
`reekystive@reeky:~/midi/lufa$ avr-size

Command 'avr-size' not found, but can be installed with:

apt install binutils-avr
Please ask your administrator.`
So I installed it
`reekystive@reeky:~/midi/lufa$`` sudo apt install binutils-avr`
But I still can't compile it
`reekystive@reeky:~/midi/lufa$ make doxygen
Executing "make doxygen" on all LUFA library elements.

make -C LUFA doxygen
make[1]: Entering directory '/home/reekystive/midi/lufa/LUFA'
 [DOXYGEN] : Configuration file "doxyfile" with parameters "QUIET=YES PROJECT_NUMBER=170418"
if ( ( cat doxyfile ; echo "QUIET=YES" ; echo "PROJECT_NUMBER=170418" ) | doxygen - 2>&1 | grep -v "warning: ignoring unsupported tag" ;); then exit 1; fi;
make[1]: Leaving directory '/home/reekystive/midi/lufa/LUFA'
make -C Demos doxygen
make[1]: Entering directory '/home/reekystive/midi/lufa/Demos'
make -C Device doxygen
make[2]: Entering directory '/home/reekystive/midi/lufa/Demos/Device'
make -C ClassDriver doxygen
make[3]: Entering directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver'
make[4]: Entering directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver/AudioInput'
 [DOXYGEN] : Configuration file "doxyfile" with parameters "QUIET=YES"
if ( ( cat doxyfile ; echo "QUIET=YES" ) | doxygen - 2>&1 | grep -v "warning: ignoring unsupported tag" ;); then exit 1; fi;
make[4]: Leaving directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver/AudioInput'
make[4]: Entering directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver/AudioOutput'
 [DOXYGEN] : Configuration file "doxyfile" with parameters "QUIET=YES"
if ( ( cat doxyfile ; echo "QUIET=YES" ) | doxygen - 2>&1 | grep -v "warning: ignoring unsupported tag" ;); then exit 1; fi;
make[4]: Leaving directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver/AudioOutput'
make[4]: Entering directory '/home/reekystive/midi/lufa/Demos/Device/ClassDriver/CCID'
 [DOXYGEN] : Configuration file "doxyfile" with parameters "QUIET=YES"
if ( ( cat doxyfile ; echo "QUIET=YES" ) | doxygen - 2>&1 | grep -v "warning: ignoring unsupported tag" ;); then exit 1; fi;`

What do I need to compile it?

abcminiuser, 2019-01-30:
Try "make" or "make all". The doxygen target just geberates the documentation for the current project, while the default "all" target does the actual build.

ReekyStive, 2019-01-30:
I successfully compiled with `make all`
Thanks!

closed 2019-01-30 by ReekyStive