abcminiuser/lufa #146: Documentation: Clarify HID_DESCRIPTOR_VENDOR documentation

cinderblock, 2019-02-10:
I am working on a project trying to implement "vendor defined" HID data. I have it working sending 8 bytes of data in two directions but I can't seem to change the payload size as I want.

I'm using the `HID_DESCRIPTOR_VENDOR` macro but I'm finding the documentation a little confusing.

I understand that HID reports in general are quick confusing and I'm not looking for an explanation of that.

Specifically, I this extract from the docs is giving me pause:
> The resulting report should be a `uint8_t` byte array of the specified length in both Device to Host (IN) and Host to Device (OUT) directions

Is the "specified length" the length of the resultant list of `USB_Descriptor_HIDReport_Datatype_t`?

How can a single length be specified in two places (IN and OUT)?

I'm also unclear on what each of the parameters does. The names `DataINUsage` and `DataINUsage` are relatively clear but not with `NumBytes` there too.

The GenericHID demo suggests `NumBytes` is actually what controls the size of the packets but I've had trouble changing that. It looks like another person has had similar confusion: https://www.avrfreaks.net/forum/generichidlufahidapivisualc-not-working-hid-records

It's also not clear if I can mess with `CollectionUsage` but I suspect that is a deeper HID spec thing.

I've tried looking at the `#defined` of `HID_DESCRIPTOR_VENDOR` but that jumble is quite confusing (as expected from the HID spec).

Any pointers greatly appreciated.

cinderblock, 2019-02-10:
Well, I stayed up getting into the USB HID specifications and tried a couple more things. I'm not sure what I was doing wrong before but I can now increase the packet size as expected.

After reading the documentation excerpt again, I believe I understand it now. Some thoughts:
> The resulting reports will be `uint8_t` byte arrays of `NumBytes` length in both directions.

The references to IN/OUT I understood to mean related to the `DataINUsage` and `DataInUsage` arguments. In my naivety on the USB spec, I did not realize "Usage" has specific meanings.

abcminiuser, 2019-02-18:
Sorry for the late reply - I've been away on a cruise ship around New Zealand, no internet access for the last week and a half.

I agree, the documentation could use some clarification. I'll mark this on the backlog for when I have a little free time.