abcminiuser/lufa #148: Atmega32u4 Mass Storage Bootloader for Pro Micro

joric, 2019-07-01:
First of all, I can't even build it now, atmega32u4/32/4 configuration never bulds even with BOARD=NONE. Maybe in 2013 it worked somehow. I'm using standard gcc toolchain on WSL (debian).

```
/usr/lib/gcc/avr/4.9.2/../../../avr/bin/ld: section .apitable_trampolines loaded at [0000000000007fa0,0000000000007fb7] overlaps section .data loaded at [0000000000007e28,0000000000007fa7]
```
My conf is:
```
MCU          = atmega32u4
ARCH         = AVR8
BOARD        = NONE
F_CPU        = 16000000
...
FLASH_SIZE_KB         = 32
BOOT_SECTION_SIZE_KB  = 4
...
```

It only builds for `BOOT_SECTION_SIZE_KB  = 6`, 4 or 5 doesn't work.

Also I can't use prebuilt Leonardo images from here http://fourwalledcubicle.com/blog/2013/03/magic/comment-page-1/ because IO13 (PC7) is not even broken out on Pro Micro. Could you please remap it to double reset or something as does Adafruit? There are very few pins broken out (just 18 gpios) and using a dedicated pin to invoke usb drive is pretty wasteful. Thanks for the great bootloader, but making it work on Pro Micros would be even greater, considering mass adoption of those boards.

joric, 2019-07-01:
Upd. Managed to build it with `BOOT_SECTION_SIZE_KB = 6` and `BOARD = NONE` using msys2 installed with [qmk script](https://github.com/qmk/qmk_firmware/blob/master/docs/getting_started_build_tools.md) (avr-gcc version is 5.4.0 now, was 4.9.2). Flashed with USBASP as so:
```
avrdude -c usbasp -p atmega32u4 -U flash:w:BootloaderMassStorage.hex
```
Just got USB drive popped on a Pro Micro (OMG!). Now I test user firmware.
Regarding IO13 and C7 I just noticed it's only for `BOARD = BOARD_LEONARDO`, `BOARD = NONE` uses different path (something with reset and fuses, not sure how it works yet).

joric, 2019-07-02:
Reopened, question of the day: why it doesn't want to overwrite old FLASH.BIN without manual deleting? Is it possible to make it work without deleting old FLASH.BIN first? STM32F103T8U6 bootloader (64K flash, 20K RAM) allows to overwrite just fine.

abcminiuser, 2019-10-27:
The bootloader only presents just enough FAT sectors to fit the application in - this is because the sectors are backed by the literal internal FLASH memory. Other MSC bootloaders generally buffer the data into RAM so that they can handle renames, long file names and other FAT features, but the AVR8 devices lack the memory to do so. Deleting the file frees up the sectors of the filesystem such that a new image can be written.

closed 2019-10-27 by abcminiuser