abcminiuser/lufa #154: INFO: LTO option clears eeprom

zygmunt, 2019-11-21:
Hi, 
something interesting to note.
Having LTO = Y will cleanup your eeprom.
I dont know how to avoid that, but would be good know how to do it.
LTO nicely optimize code, but wiping out eeprom may brake few of your days:)

NicoHood, 2021-01-29:
What do you mean by cleanup? When does it happen? On every boot, when flashing the device? This sounds strange to me.

arunmjayan, 2021-01-30:
Recently i noticed that the chip manufacturers only are giving us the HID
example code for that chip, for example since i was working with a adafruit
board they only had the HID code, which got after a small digging.
From what i understood from that code is that, instead of using c for the
RawHID communication, you could write a small python program importing the
hid package from the python library itself
import hid.
Thank You. Regards