abcminiuser/lufa #164: teensy-ee targets fail to build for BulkVendor, VirtualSerial, and others

khimaros, 2020-07-06:
this is happening at 790ac4d125c1c5d72016b52c2b9ad58f5e5f8c62

reproduction steps:

```
$ pwd 
[...]/lufa/Demos/Device/LowLevel/VirtualSerial
$ make teensy-ee
 [OBJCPY]  : Extracting HEX file data from "BulkVendor.elf"
avr-objcopy -O ihex -R .eeprom -R .fuse -R .lock -R .signature BulkVendor.elf BulkVendor.hex
 [OBJCPY]  : Converting "BulkVendor.hex" to a binary file "InputEEData.bin"
avr-objcopy -I ihex -O binary BulkVendor.hex ../../../../LUFA/Build/DMBS/DMBS/HID_EEPROM_Loader/InputEEData.bin
 [MAKE]    : Making EEPROM loader application for "BulkVendor.hex"
make -s -C ../../../../LUFA/Build/DMBS/DMBS/HID_EEPROM_Loader/ MCU=at90usb1287 clean teensy
 [RM]      : Removing object files of "HID_EEPROM_Loader"
 [RM]      : Removing dependency files of "HID_EEPROM_Loader"
 [RM]      : Removing output files of "HID_EEPROM_Loader"
make[1]: *** No rule to make target 'obj/InputEEData.o', needed by 'HID_EEPROM_Loader.elf'.  Stop.
make: *** [../../../../LUFA/Build/DMBS/DMBS/hid.mk:55: teensy-ee] Error 2
```

khimaros, 2020-07-06:
i'm also running into this with the 170418 release.

khimaros, 2020-07-06:
i've been able to work around this with the following steps:

```
$ cd <my project dir>
$ make teensy-ee
[...]
make[1]: *** No rule to make target 'obj/InputEEData.o', needed by 'HID_EEPROM_Loader.elf'.  Stop.
make: *** [../../../../lufa/LUFA/Build/DMBS/DMBS/hid.mk:55: teensy-ee] Error 2
$ cd ${LUFA_PATH}/LUFA/Build/DMBS/DMBS/HID_EEPROM_Loader/
$ avr-objcopy -I binary -O elf32-avr -B avr5 --rename-section .data=.progmem.data,contents,alloc,readonly,data InputEEData.bin obj/InputEEData.o
$ make teensy
```

it seems that there is some sort of conflict between the implicit `%.o` targets defined in gcc.mk and the `InputEEData.o` target defined in `HID_EEPROM_Loader/makefiles`

khimaros, 2020-07-06:
for context, this is with GNU Make 4.2.1.

after a bit more poking around, the following patch works for my case:

```
diff --git a/LUFA/Build/DMBS/DMBS/HID_EEPROM_Loader/makefile b/LUFA/Build/DMBS/DMBS/HID_EEPROM_Loader/makefile
index 879eda8cf..12879eac8 100644
--- a/LUFA/Build/DMBS/DMBS/HID_EEPROM_Loader/makefile
+++ b/LUFA/Build/DMBS/DMBS/HID_EEPROM_Loader/makefile
@@ -22,12 +22,12 @@ OBJECT_FILES = InputEEData.o
 all:
 
 # Determine the AVR sub-architecture of the build main application object file
-FIND_AVR_SUBARCH = avr$(shell avr-objdump -f $(TARGET).o | grep architecture | cut -d':' -f3 | cut -d',' -f1)
+FIND_AVR_SUBARCH = avr$(shell avr-objdump -f obj/$(TARGET).o | grep architecture | cut -d':' -f3 | cut -d',' -f1)
 
 # Create a linkable object file with the input binary EEPROM data stored in the FLASH section
-InputEEData.o: InputEEData.bin $(TARGET).o $(MAKEFILE_LIST)
+obj/InputEEData.o: obj/HID_EEPROM_Loader.o
        @echo $(MSG_OBJCPY_CMD) Converting \"$<\" to a object file \"$@\"
-       avr-objcopy -I binary -O elf32-avr -B $(call FIND_AVR_SUBARCH) --rename-section .data=.progmem.data,contents,alloc,readonly,data $< $@
+       avr-objcopy -I binary -O elf32-avr -B $(call FIND_AVR_SUBARCH) --rename-section .data=.progmem.data,contents,alloc,readonly,data InputEEData.bin $@
 
 # Include LUFA build script makefiles
 include ../core.mk
```

the build steps i tested:

```
$ cd <my project dir>
$ make teensy-ee
<press bootloader button>
$ make teensy
```

digging further; renaming the `InputEEData.o` target to `obj/InputEEData.o` isn't sufficient, as the dependency on `InputEEData.bin` seems to cause a transitive cyclical dependency via the `%.bin` target in gcc.mk:

```
[...]
        Considering target file 'InputEEData.bin'.
         Looking for an implicit rule for 'InputEEData.bin'.
         Trying pattern rule with stem 'InputEEData'.
         Trying implicit prerequisite 'InputEEData.elf'.
         Trying pattern rule with stem 'InputEEData'.
         Trying implicit prerequisite 'InputEEData.elf'.
         Looking for a rule with intermediate file 'InputEEData.elf'.
          Avoiding implicit rule recursion.
          Trying pattern rule with stem 'InputEEData'.
          Trying rule prerequisite 'obj/InputEEData.o'.
          Trying rule prerequisite 'obj/HID_EEPROM_Loader.o'.
         Found an implicit rule for 'InputEEData.bin'.
make[1]: Circular InputEEData.elf <- obj/InputEEData.o dependency dropped.
 [GCC]     : Compiling C file "HID_EEPROM_Loader.c"
 [LNK]     : Linking object files into "InputEEData.elf"
obj/HID_EEPROM_Loader.o: In function `main':
HID_EEPROM_Loader.c:(.text.startup.main+0x4): undefined reference to `_binary_InputEEData_bin_size'
HID_EEPROM_Loader.c:(.text.startup.main+0x6): undefined reference to `_binary_InputEEData_bin_size'
HID_EEPROM_Loader.c:(.text.startup.main+0xe): undefined reference to `_binary_InputEEData_bin_start'
HID_EEPROM_Loader.c:(.text.startup.main+0x10): undefined reference to `_binary_InputEEData_bin_start'
collect2: error: ld returned 1 exit status
make[1]: *** [../gcc.mk:250: InputEEData.elf] Error 1
make: *** [../../../../lufa/LUFA/Build/DMBS/DMBS/hid.mk:55: teensy-ee] Error 2
```

abcminiuser, 2020-07-19:
Thanks for looking at this. It's actually an issue in the upstream DMBS project, which I import into LUFA here. It's also trying to embed the application binary rather than the EEPROM data for the Teensy target, which is doubly embarrassing.

I've fixed both in https://github.com/abcminiuser/dmbs/commit/3f76496f8d12c3e7d52e15b8d958d8fd0faf68a8 and imported it here in https://github.com/abcminiuser/lufa/commit/305ca461acf37a5a2237afe7d2ce552ce25075a5 .

The CI build failure was a dumb mistake, corrected in https://github.com/abcminiuser/lufa/commit/ca40bc7a6cca203678f122c95c0ab7a775f8497a .

Please-reopen if it's still failing for you, but I see it generating and attempting to program the EEPROM loader application when testing the updated code locally.

closed 2020-07-19 by abcminiuser

khimaros, 2020-07-19:
This worked for me. Thank you, Dean!