abcminiuser/lufa #167: Implementación LUFA a MPLAB X

Borxo, 2020-09-14:
Hi, My is Borja
I have a question.
Has anyone implemented the LUFA library in Mplab X? Is it possible to use it in Mplab X?

Since it seems that Microchip is creating the tools for all its microcontrollers.

Thank you very much

abcminiuser, 2020-09-14:
Hi Borxo,

I don't currently have any plans to do this - I'm not even sure what 
sort of extension/integrations are possible with MPLABX. You should be 
able to create a regular project and copy in the relevant source files 
however, it's just the module system of Atmel Studio that would be missing.

- Dean

Borxo, 2020-09-15:
Thank you very much for the information.
Follow their guidelines to use it at Atmel Studios.
I close the comment.

closed 2020-09-15 by Borxo