abcminiuser/lufa #169: LUFA extension Atmel Studio 7 - update for version 210130

Principia4834, 2021-02-15:
Dean
It looks like the LUFA extension for Atmel / Microchip Studio 7 is not using LUFA 210130. Would it be possible to add this update to your TODO list?
Andy

abcminiuser, 2021-02-24:
I actually stripped out all the Atmel Studio support from the repository; it was hard to maintain, and since the rebranding to Microchip Studio I lost access to the extension on the Atmel/Microchip Gallery, so I couldn't update it even if I wanted to.

The current direction seem to be towards MPLAB-X, and the Atmel Software Framework that I used for the LUFA integration was effectively deprecated for ATMEL Start.

Principia4834, 2021-02-24:
Ok - thanks for the update. I've grabbed the 2017 extension files and I'll probably build for myself.

closed 2021-02-24 by Principia4834