abcminiuser/lufa #180: LUFA USB CDC - connection with serial application fails after disconnect and reconnect

UriyMihailov, 2021-12-27:
Hello, I wanted to ask a question about a problem I can't find a solution to. I used a Virtual Serial demo with Atmega32U2, made a few modifications so it could receive and send bytes and it works like a charm. However, whenever I unplug the device and plug it in again, the device stops both receiving and sending info to the PC application, although the application still sees the Virtual COM the device is assigned to. I tried searching for info about this problem at different forums, and it usually says that it could be a Windows problem, it's drivers problem or even the PC application problem, so there is potentially nothing you can do about it. But a friend of mine says that this is unlikely because he used that specific application and some of the devices reconnected correctly, while others had problems. So I wanted to ask you whether you could guide me to the possible solution to this problem, IF it could be solved through alterations of the demo's code?

abcminiuser, 2022-01-16:
Hi Urly, did you ever figure this one out?

USB to Serial adapters on Windows can use of of several protocols; the CDC-ACM that is implemented by LUFA, and/or proprietary protocols such as those used by FTDI. The CDC-ACM standard driver in Windows has trouble with port naming; if a port is open when a device disappears and re-appears, it will (usually) get assigned to a new and unused COM port number.

That's quite annoying behavior as it means on Windows CDC-ACM serial ports don't have stable port names, but as far as I am aware there isn't a good workaround. Third party proprietary drivers and protocols work around this issue, but aren't documented thus aren't implemented in LUFA currently.

UriyMihailov, 2022-01-19:
Yeah, sorry, I couldn't find an appropriate solution to this. Seems like it is a PC application problem, so for the time being I'll just have to bear with it or write my own app. Anyway, thank you for the info you provided. I think the issue may be closed for the moment.