abcminiuser/lufa #184: Unknown reference to "UpdateCurrentCompatibilityMode"

marecl, 2022-09-11:
Hi
I'm currently messing around with AVRISPmkII code and I found that function ```UpdateCurrentCompatibilityMode()``` is not defined anywhere.  
I've searched through all LUFA and AVR sources but cannot find anything on this function. It doesn't exist in LUFA docs and sources, AVR toolchain or project files.
Where is this function declared? How come VSCode can't compile it but AVR Studio can?

Thanks in advance

abcminiuser, 2022-09-11:
Should be fixed in eb9b67b23ef02e7b30fec26eed7c56b5630b9682 - there was some ancient vestigial code in there that should have been deleted, for a compile time option that was removed years ago. Not sure why your IDE would be setting that old pre-processor definition that would call it however.

marecl, 2022-09-12:
Thanks for your help! :)
I enabed it manually. Together with some other options but this one has been giving me headaches  
While searching I also found it in XPLAINED Bridge.

closed 2023-09-19 by abcminiuser